opencpn-radar-pi/radar_pi #1: Garmin radar packet capture files needed

canboat, 2017-10-09:
If you have a Garmin HD, xHD or Fantom radar we're interested in `.pcap` files.

To create one on Linux, use `tcpdump`. To create one on Microsoft Windows, use a tool such as ''Wireshark''.

To be really useful we'd like a number of files, preferably from people with a Garmin plotter.

## Type 1 - power and zoom

* Starting the radar from the plugin or Garmin chart plotter
* Zooming out to the largest range.
* Zooming back in to the smallest range.

## Type 2 - controls
* Change the gain / sea state etc from minimum to maximum, then select auto (if it exists.)

Create a separate file per control, and make sure you name it so that it can be recognised what control you were changing.

You can upload the files here in this issue. Please zip them, as they become a lot smaller that way.

canboat, 2017-10-30:
Hans has sent me files for Garmin xHD. Still looking for HD and Fantom files.

hughespat57, 2017-11-30:
I just got hold of a Garmin GMR18HD and would be glad to provide.
However I cant get it up and running correctly yet.
WireShark sees multicast boot up message when I power it on, and then a stream of multicast messages.
OpenCPN sees it, joins the multicast and I can even see data going back to the 18HD if I change various settings in the Radar Controls
But OpenCPN never fully initializes the radar, control icon red, and no button functionality to "turn on" unit.
From a network view all seems good, may have a hardware issue with the GMR, does the plugin pickup Garmin error messages anywhere? I dont have access to a Garmin ChartPlotter to test it or upgrade the GMR firmware right now.

trudK45, 2017-11-30:
Do you have a GarminHD or a GarminxHD radar ?
I expect you have the HD since you can't get it running.
This plugin only handle the xHD version yet.
If you have the HD version I am sure the team would be very happy if you could be a test pilot for this radar even if it may have to wait a little while since Kees may want to work of some bugs first.
Kees will surely come back to you.

canboat, 2017-11-30:
The HD version is even simpler than the xHD so it won't be very difficult
to port the code over from the original gradar plugin. I will probably do
this in the next weeks.

Do note that the new plugin is a work in progress so it helps if you are
able to do light debugging. We may ask you to tinker with some code to get
it working, once the majority of the code is in, so getting a development
environment running as explained in the opencpn developer documentation is
very useful.

canboat, 2017-11-30:
For now you should probably also verythat your radar works with the
original 'gradar' plugin, if it is a HD and it doesn't work with that it
will also not work with this plugin unless you find out what the problem
is. Since i don't have Garmin radar(s) there is only so much I can do.

trudK45, 2017-11-30:
I have copied the xHD code into a new directory under src, changed every instance of xHd to HD , built the code and tested it against my radar with full functionality. I am about to change the code so it will work with the HD radar. Is it ok with you Kees if I commit at this stage ?

canboat, 2017-11-30:
Sure go ahead. Did you change the control messages to the ones desired by
the HD radar?

trudK45, 2017-11-30:
Not yet. Next to do. There has been a lot added and changed so want to get this up first. We have been delayed so still another week or two with good Internet.

canboat, 2017-11-30:
The best way to do something like this is to do this on a separate branch
until the feature is ready, but since we are nowhere ready for a release
yet it is acceptable if you do this on master.

hughespat57, 2017-11-30:
Sorry let me be clear, I cant get it to function yet with the 1.5 gradar_pi
So till I get that resolved I may be of no value...

trudK45, 2017-11-30:
You can see the packages is WireShark if understand it right.
Have you activated the log window ?
If so do you get this message, "->gradar_pi: First Packet Rx" ?
Maybe the firewall is preventing the data to get to the plugin.

hughespat57, 2017-11-30:
Yes I get First Packet, unfortunately I think the network is good, thus I may be facing a hardware issue in the GMR18HD logic board. These were know to have some issues early on.
I am not a coder, but I am a computer systems engineer, so quite familiar with WireShark and digital systems. I have taken everything down to the basics, direct connect network, firewall disabled on this interface. I can ping the scanner even. 
     0: Found network interface: 172.16.2.200/24
     1: Found network interface: 192.168.13.23/24
     2: Found network interface: 127.0.0.1/8
     3: Scanner at 172.16.2.0 is accessible.
     5: UpdateState:  PluginState  [M]  Searching  (1)   Scanner state:  Unknown  (0)
     4: Successfully added to multicast group 239.254.2.0
     6: ->gradar_pi: First Packet Rx     
     7: UpdateState:  PluginState  [M]  Searching  (1)   Scanner state:  Unknown  (6)
     8: Scan packets per tick: 2
     9: True Heading:  -1.#IND  Mag Heading:  -1.#IND  Variation:  -1.#IND    thpri: 99  mhpri: 99  vpri: 99
    10: True Heading:  0  Mag Heading:  -1.#IND  Variation:  -7.53052    thpri: 99  mhpri: 99  vpri: 5

hughespat57, 2017-11-30:
My only hope was/is that somehow its not initializing completely and I cant get gradar_pi to let me send a TX command, as the buttons are never active. Funny as I exit OpenCPN it sends a final "TX Off", even though its not on, a good code bit to be sure the sensor shuts down I assume.

trudK45, 2017-12-01:
Are you able to upload a .pcap file to your github or a dropbox ?

trudK45, 2017-12-01:
Does anyone know the scan lengths for the Garmin HD radar ?

hughespat57, 2017-12-01:
[Garmin18HD-OpenCPN gradar_p1 1.5 log-with notes-PJHughes.txt](https://github.com/opencpn-radar-pi/radar_pi/files/1522406/Garmin18HD-OpenCPN.gradar_p1.1.5.log-with.notes-PJHughes.txt)

hughespat57, 2017-12-01:
[cold start-w-radar controls.zip](https://github.com/opencpn-radar-pi/radar_pi/files/1522411/cold.start-w-radar.controls.zip)

hughespat57, 2017-12-01:
The only advantage I have right now is this is set up as a lab in my office.
So I can do anything at any time...
Running off an 80 Amp 12.2 volt regulated bench power supply and a dedicated laptop
Windows 10 64 bit, Lenovo W700

trudK45, 2017-12-01:
For some reason the radar is stuck in the warmup state.  The radar state is at 0006 and the warmup timer is at 0001 for every response packet and does not ever go to zero. This means the the radar symbol will be blank, It should go to amber when the radar has warmed up.
There seem to be something wrong here. Do you have pin 5 of the F-F Ethernet connector grounded ?

hughespat57, 2017-12-02:
Yes i have the pin 5 connected to ground
And it does react if i cycle that  

I have been probing the unit over the last few days and am focused now on the the magnetron heater so this makes sense.
The power points all come up during the initial self test phase. I get 350 volts on the HV primary momentarily then turns off, don't have a high voltage probe for checking the secondary 3kv output. Dangerous stuff magnetrons: HV, microwaves and xrays.
But not getting the 6 volt heater current for the magnetron, that seems to be the problem
Until that heats up for 60 seconds minimum it will never allow the HV transmit voltage

I am bidding on a beat up garmin 3010c plotter if i get it I can flash the firmware up to date and pull error codes

Plus then i'll have a real chart plotter to do some packet traces on
 
Patrick

trudK45, 2017-12-02:
Kees!
I have modified the code to match the Garmin HD radar. There are issues still with some of the commands and status messages but I believe the code will recognize the radar.
I think there are menus that need to be modified and I am not yet ready to do this.
Further more I have no way of testing.
We need someone with a HD scanner.

canboat, 2017-12-02:
Cool. 

Yes at the very least we need some pcap files of someone with a working radar.

I am going sailing hopefully later today (if the fog clears …) and then I can look at the code at anchor.

trudK45, 2017-12-02:
Oh, and forgot to tell you, I have not done anything with the scan data as I do not know the The spoke length. There are 720 pulses per revolution and each packet contains 4 spokes. I thing the received data is pure binary. I.e. one byte contains 8 slots. Maybe you know more about this.

liquidus35, 2018-10-26:
Hello,
Wondering if there is still interest in some captures from a Garmin Fantom radar?  I just installed a Fantom last month and will have access to it again in about a week.  If there is interest, is there anything special I need to do to capture the packets with wireshark?  I have not had a chance to attach my laptop up to the Garmin GMS 10 hub to see what the network interface looks like from my PC.  Should I assume the my Garmin MFD acts as a DHCP server?  Or maybe the radar does DHCP?  I know my way around networks pretty well but am a complete newbie to how Garmin configures its marine network.   So any guidance or links to how to get setup would be helpful.

Eventually I want to get OpenCPN running at my Nav station below on a PC but for the moment I would like to learn more about the Garmin network and what data might be extractable from it.  But if I can contribute to getting the Fantom radar on OpenCPN I would be happy to do so.  

Cheers

trudK45, 2018-10-27:
I am sure we want Wireshark recordings for your radar. I don't know about the Phantom but on the HD and xHD (the one I have)  this document applies [https://opencpn.org/OpenCPN/doc/GRadarDoc.pdf](url) 
Especially note: 
For operation of the PlugIn in Master mode, additional configuration is required.  The general idea is to configure
an Ethernet port on the OpenCPN computer with an IP address belonging to the group in which the radar scanner listens and
resides.  **This group address is hardwired into the radar scanner**, and is the private (non­routable) address range
172.16.0.0/24.  Formally, we may choose any specific IP address within this group for the OpenCPN computer, other than the
address of the scanner or the chartplotter. Practically, the OpenCPN Radar Plugin assumes that the IP address will be set to
172.16.1.1/24.

keesverruijt, 2018-10-27:
Hi Liquidus, 

Welcome. Our experience shows that we need someone with some development skills and direct access to the radar to develop a successful plugin if it is a new radar type. 

The network recording will tell us whether the Fantom is a "better xHD" or a different beast altogether. We just don't know. 

If it is significantly different we depend on someone to step up, take the existing work, copy the closest radar, and start developing. We're happy to help anyone in their first steps (defining the data structures that add a new radar type to the plugin.)

liquidus35, 2018-10-29:
Thanks for the config info trudK45.  That should get me started.  I will see what I can collect when I'm down at the boat in the next 2 weeks.  I won't have a chance to post anything until after 15 November most likely.  

Another related question is whether anyone has looked at recording NMEA 2000 traffic that occurs as you change the Radar settings on the MFD.  I tried some basic tests when I was at the boat the last time and noticed that I could not operate the Garmin Fantom radar from the Garmin MFD unless I had both NMEA 2000 and the Garmin Marine network (ethernet) connected to the MFD.  So I'm guessing there is data being passed by NMEA 2000 as well as ethernet.  
Can anyone confirm this?  

Does anyone have a suggestion for a low cost NMEA2000/CAN bus/Devicenet to PC USB converter? 

How about a Wireshark-like program for NMEA 2000 traffic capture?

keesverruijt, 2018-10-29:
> Another related question is whether anyone has looked at recording NMEA 2000 traffic that occurs as you change the Radar settings on the MFD. I tried some basic tests when I was at the boat the last [...]
First time I’ve heard this. But there aren’t that many Fantom radars around. Does NMEA 2000 go into the Fantom? The installation manual only mentions “Garmin Marine Network” but that sounds like Ethernet.

What it could be is that the MFD wants to have heading before it does anything with the radar. Try it on a real “radar page” without chart overlay.

> Does anyone have a suggestion for a low cost NMEA2000/CAN bus/Devicenet to PC USB converter?
> 
METS should bring some news for you.
> How about a Wireshark-like program for NMEA 2000 traffic capture?
> 
> 

https://github.com/canboat/canboat

… also by yours truly…

Kees

liquidus35, 2018-10-29:
I stand corrected.  I did not perform the Fantom install and just assumed from looking at the connector that the radar was NMEA 2000 powered.  But as I mentioned, the MFD doesn't seem to recognize the radar without the NMEA 2000 connected.  At this point don't give this observation too much further thought until I have a chance to verify again.

canboat looks perfect!  Thanks Kees!  I am still interested in capturing NMEA 2000 data to understand how the Raymarine IC-5 analog converter module gets its calibrations programmed from the i70 display.  Raymarine claims you have to have a Raymarine i70 display in order to send calibrated wind, depth, temperature, compass, and rudder data from the IC-5 converter over the NMEA 2000 network.

liquidus35, 2018-11-15:
Update:  A bit of a delay in getting Wireshark data from the Fantom radar system.  Ended up sailing the whole time  and stayed off the computer this trip.  :-)   I will try again during the next trip in a couple of months. Sorry for the false start.

tyguybye, 2019-08-21:
Not sure if this old thread is still active, I have a Garmin GMR 18 Fantome running in home office setup through RPI or Windows running OCPN.  I do not have access to a MFD but can take wireshark data if this will help, I am available 24/7 for the next 2 weeks if that is of any help.

Hakansv, 2019-08-21:
@tyguybye 
Support for any broadband radar would be possible with radar_pi. The code is also adapted to as much as possible be useful on any radar scanner. 
The work to program any new scanner is to capture commands and replies to and from the scanner. For a Wireshark capture you need to send a known command from a MFD or another program that can do the communication. 
See also the similar topic and Douwe's advise here: https://github.com/opencpn-radar-pi/radar_pi/issues/116
When all commands are captured next step is to program for that radar scanner. This is not a self playing piano of any kind. But the first and important step is to identify and capture commands and other settings. Next step is to interpret the captures and try to write the first code for test.
All support and participation to achieve this is of course appreciated.

tyguybye, 2019-08-21:
@ Hakansv I thought that might be the case, I was just going to throw my hat in the ring for an older topic that I thought I might be able to contribute to. I wish I had the MFD to send those commands, oh well I will wait for someone that is able to get the awesome Fantom radars to work in OCPN.

mmfriedel, 2020-02-14:
Here we go. Let me know what else is needed.
[Fantom18Capture.zip](https://github.com/opencpn-radar-pi/radar_pi/files/4207122/Fantom18Capture.zip)

mmfriedel, 2020-02-14:
Just as additional comment. With option set to xHD some things work. It detects a radar, I can for example turn on/off transmit. But nothing is displayed. Il spin up the debugger in the near future and take a closer look.

trudK45, 2020-02-15:
@mmfriedel I had a quick look at your captured files. From what I can see packet lengths differ from the xHD. The announcement package found in StartupTransmit  has the same id value of  (0x0392). 
The Radarvideo packages seem to have the same id (0x0998). Packet length different. I think you need to take a step back and start with the commands and make one wireshark capture for each command so they are easily detectable in the data stream.  Don't start transmitting at this point.  When all commands are identified and decoded we can continue with the status messages returned from the radar dome.  /Hans

mmfriedel, 2020-02-15:
O.K. I will turn off the transmit… I did send only one command for every capture. 
Some commands are only accessible when transmit is on though.

The display not working was a user error (did not have accelerated graphics on).
I played with it a little. Most commands work, display also works (Window and Overlay). 
What doesn’t work is dual range.. and I did not have the chance to play with the ‚doppler‘ features. 
I tried changing range gain/mode/setting no transmit zones. That all mostly worked. Some ranges didn't.
I will take a more systematic approach, and make separate capture files.

Tested

Transmit On/Off
Change Range
Change Gain
Change Gain Mode
Set No Transmit zones

Paket length changes based on range.
Number of Spokes seems to be the same (1440).

> @mmfriedel <https://github.com/mmfriedel> I had a quick look at your captured files. From what I can see packet lengths differ from the xHD. The announcement package found in StartupTransmit has the [...]

Michael M Friedel
mfriedel@ntropy.org

trudK45, 2020-02-16:
@mmfriedel Are you able to build radar_pi ? I.e. do you have the development environment set up ?
/Hans

mmfriedel, 2020-02-16:
Yes. I got the build environment to work.

BTW do you have a dissector for Wireshark to look at the packets ?

> 
> @mmfriedel <https://github.com/mmfriedel> Are you able to build radar_pi ? I.e. do you have the development environment set up ?
> /Hans

Michael M Friedel
mfriedel@ntropy.org

mmfriedel, 2020-02-17:
I have made a copy of garminxhd as garminfantom. Once I get back to the boat I will report if that worked.
I have a GMR Fantom 18 (but there is also a 24,54 and 124). So I figure there is probably some kind of query in the beginning that reports the type. 
Plans is to turn on chart plotter and capture traffic.
And given that the xHD and Fantom are on the same ports.. there should probably be should be a check.

> 
> @mmfriedel <https://github.com/mmfriedel> Are you able to build radar_pi ? I.e. do you have the development environment set up ?
> /Hans

Michael M Friedel
mfriedel@ntropy.org

trudK45, 2020-02-17:
Is it the standalone garminxhd or the radar_pi you made a copy of ?
I suggest you make a fork of the existing radar_pi and go from there if that's not the case.
I do not have something that I cab use to decode the packets. I manually decode the packets and adjust the code for each command/response.
Here is a document describing the packets.
[https://github.com/opencpn-radar-pi/radar_pi/tree/master/doc](url)
/Hans

keesverruijt, 2020-02-25:
If you are able to compile, that's a great step up.

Fork the project, then copy the Garmin xHD files and name them "Fantom" or
"Fantom18". Then you can start modifying to support the Fantom improvements
over xHD.

- You'll need to check that the radar doesn't send out more data if the
radar spokes are a different length.
- The ranges are probably easy to fix if you have access to the MFD, just
let the chart plotter send out each range command and observe this using
Wireshark.

Dual range and Doppler *could* be simple, especially if they implemented it
the same way as Navico's HALO. Navico uses different addresses for radars A
and B, if that is the case for the Fantom then simply do as what we've done
for Halo and 4G: define two radars, with the different addresses (but
sharing the rest of the code, obviously...) Doppler is most likely to be
transmitted like HALO: special values reserved for the Doppler colours.
Anything else just doesn't make sense!

Good luck, and thanks for the help,

Kees

mmfriedel, 2020-02-25:
I have done, so I have also made some changes. I did not have a chance to get back to the boat yet.
Doppler.I plan to record with Wireshark while observing a doppler target. Then I will check the values.
Not sure how they have done the dual range, but I will check. I figured either different spoke numbers or a different address.
Mostly I am trying to find how the MFD queries the Radar Type. Would be nice to make the discovery automatic. 

Also thinking of adding a feature to record any unknown status messages with some stats. Might be useful to help implement all the radars one doesn't have access to, that way people willing to help don't have to compile anything or mess with the command line or wireshark.

> If you are able to compile, that's a great step up.
>
> Fork the project, then copy the Garmin xHD files and name them "Fantom" or
> "Fantom18". Then you can start modifying to support the Fantom [...]

Well so far they have not hit the max length defined. They do change length depending on range.

> - The ranges are probably easy to fix if you have access to the MFD, just
> let the chart plotter send out each range command and observe this using
> Wireshark.
>
> Dual range and Doppler *could* be [...]

Ill keep you posted. I’ll push the changes once I have confirmed they work.
Thank you for the great work, the changes necessary are pretty minimal.

The Fantom seem to be a very close cousin to the xHD. Apart from some ranges not working, most commands are the same.

> 
> Kees
> 
> 
> > Is it the standalone garminxhd or the radar_pi you made a copy of ?
> > I suggest you make a fork of the existing radar_pi and go from there if
> > that's not the case.
> > I do not have something that I cab use to decode the packets. I manually
> > decode the packets and adjust the code for each command/response.
> > Here is a document describing the packets.
> > https://github.com/opencpn-radar-pi/radar_pi/tree/master/doc <http://url>
> > /Hans
> >
> > I have made a copy of garminxhd as garminfantom. Once I get back to the
> > boat I will report if that worked. I have a GMR Fantom 18 (but there is
> > also a 24,54 and 124). So I figure there is probably some kind of query in
> > the beginning that reports the type. Plans is to turn on chart plotter and
> > capture traffic. And given that the xHD and Fantom are on the same ports..
> > there should probably be should be a check.
> > On Feb 16, 2020, at 5:14 PM, trudK45 *@*.***> wrote: @mmfriedel
> > <https://github.com/mmfriedel> https://github.com/mmfriedel Are you able
> > to build radar_pi ? I.e. do you have the development environment set up ?
> > email directly, view it on GitHub <#1
> > <https://github.com/opencpn-radar-pi/radar_pi/issues/1>?email_source=notifications&email_token=AENE62XASIDPIZI3WQU2MPTRDG3EPA5CNFSM4D6LEOKKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEL4UCQA#issuecomment-586760512>,
> > Michael M Friedel mfriedel@ntropy.org

Michael M Friedel
mfriedel@ntropy.org

keesverruijt, 2020-02-25:
Any help is much appreciated!

mmfriedel, 2020-02-29:
So short update. Dual range is not implemented with two target addresses. It sends reports/data with different packet types.For example the range report/command is  0x981e and 0x981fThe actual spoke data is mostly differentiated by different ranges in the spoke , and one of the fill fields (fill3 byte 0)packet type | length      | fill1 | slen  | angle | fill2 | range       | display     | fill3 | slb   | fill4 | sli         |       |       |            | 956         |       | 956   |       |       | 7408        | 7424        |       | 928   |       |             |       |       |98 09 00 00 | bc 03 00 00 | 01 01 | bc 03 | 18 1c | 00 00 | f0 1c 00 00 | 00 1d 00 00 | 00 01 | a0 03 | 08 01 | a0 03 00 00 | 00 00 | 00 00 |98 09 00 00 | d2 02 00 00 | 01 01 | d2 02 | 10 1c | 00 00 | 80 b6 02 00 | 50 5b 01 00 | 01 01 | b6 02 | 08 01 | b6 02 00 00 | 00 00 | 00 00 |            | 722         |       | 722   |       |       | 177792      | 88912       |       | 694   |       |             |       |       |Not sure one can listen two the same address twice.. so need to figure a way out to share receiving instance.Next I will figure out which commands/modes are different.

maxiu123, 2020-04-16:
Hi there!
I am willing to help with Fantom. I have now mounted Fantom and Garmin7400. I will do whatever You need to help You with this. I have PC onboard with Windows, but I can connect linux. 
I am obout to connect plotter, and the quest. is If I should connect it with normal or crossed cable? Can I connect to the switch/router without problem? I I have my PC connected to internet and Team Wiever onboard, so You can even do it Yourself :). Tell me what you need, and I will do!
regards 
Max

KastB, 2020-05-26:
Hi,

I would be really interested in a version of this plugin that can handle the Fantom devices.
As soon as I can put my hands on such a device I would try to improve the plugin on that as well.
@mmfriedel did you develop your improvements in a separate fork/branch, such that I could have a try?
Even if it is "just" an intermediate state, it would be a great starting point from me. Maybe we could even cooperate with the development.

Cheers
Bernd

keesverruijt, 2020-05-26:
Great to see progress being made on Fantom!

mooselips, 2021-01-05:
I have a Garmin GPSMap 7612 with Fantom 24 and a Windows 10 PC running OpenCPN here in my office.  I am ready to roll.

I did not wire up the Fantom yet, but I have the GPSMap up and running along with the computer.  I just need to get the network connection setting right and I can start pulling data also. (So far, my attempts at using the destination IP's and ports I found from the Wireshark Scan did not work.  I am new to OpenCPN networking.  In the past I only used it on a laptop with a USB puck).  I will attach my Wireshark scan if anyone can send me a note to configure the Network Connection.
[GPSMap 7612.pcapng.pcapng.gz](https://github.com/opencpn-radar-pi/radar_pi/files/5767804/GPSMap.7612.pcapng.pcapng.gz)

I am happy to help here.

> I have made a copy of garminxhd as garminfantom. Once I get back to the boat I will report if that worked. I have a GMR Fantom 18 (but there is also a 24,54 and 124). So I figure there is probably [...]

trudK45, 2021-01-05:
@mooselips The hookup should be as described in this document
[https://opencpn.org/OpenCPN/doc/GRadarDoc.pdf](url) 
One could argue this should be under radar_pi as well.
I think you want Interface type 1.
Today it's difficult to get hold of a hub. You would want a switch with a mirroring port and that port connected to the computer to see all the data on all ports of the switch,
Something like this : [https://www.amazon.com/TP-LINK-TL-SG105E-5-Port-Gigabit-Version/dp/B00N0OHEMA/ref=sr_1_3?dchild=1&keywords=switch+port+mirroring&qid=1609816072&sr=8-3](url)
Unfortunately this one is 5V. You would want a 12V powered switch.
I do not have access to my boat so I can't give you the brand of the one I have which is 12V.
Read the document and come back with questions.

trudK45, 2021-01-05:
@mooselips I think it's this one I got.
[https://www.amazon.com/NETGEAR-5-Port-Gigabit-Ethernet-Managed/dp/B07PJ7XZ7X](url)
Anyhow it's 12V and got mirroring.
Oops, I closed the thread. Opened it again.

trudK45, 2021-01-05:
@mooselips I now saw that the hardware description is located also under radar_pi  Wiki.
[https://github.com/opencpn-radar-pi/radar_pi/wiki/Hardware-installation-%28Garmin%29](url)

mooselips, 2021-01-05:
@trudK45 

Thank you for the link to the switch.  I will also check at work tomorrow (I am the tech teacher at a small K-8 school).  I think I put a bunch of small hubs in a pile to throw out that would at least work for testing also.  I can order the switch you listed also, but things have been slow from Amazon where I live.  I will hopefully find a hub tomorrow at work and will make some ethernet cables that I can pull pin 5 from.

Even so, shouldn't I be able to see position and other data from the ethernet out of back of the plotter?  I cannot get OpenCPN to see any sentences from ethernet and I was just worrying that I am doing something wrong on the simple bits before I get to the more complicated bits.

trudK45, 2021-01-05:
@mooselips
I suppose you are talking of the GPS data. This data is not present together with radar data.
(At least this is how it is on the xHD radar)
This data needs to be entered into the PC via a COM or USB port.
Then it can be transmitted onto the ethernet port by settings in OpenCPN and thus be recorded. This is useful especially if you have a heading compass connected for radar overlay. This would enable you to have synchronized data when running from the Wireshark data for debugging purpose. You or I can then run the whole scenario without a connected radar. All data needed is present. What you can¨t do of course is to send control data (pushing buttons). This comes later. You should start with figuring out what all the commands looks like. Then the responses of those commands. I.e. the change in status coming back as a response to the commands. You work your way up by making control data from OpenCPN working letting your plotter display the radar picture. Radar data into OpenCPN is taken care of as the last task.
This is how I did the reverse engineering for the Garmin xHD radar.
You can do as much of the this work as you feel like and I or some of the other's here help with the rest.
If you are interested, have a look at this document ; [https://github.com/opencpn-radar-pi/radar_pi/blob/master/doc/GarminEthernetData.ods](url)
It describes the packages for the xHD radar.

mooselips, 2021-01-05:
Thank you.  The GarminEthernetData.ods helps me see what I am looking for more.  I will definitely need some help with this.

Late tonight I will try to get the full system together and ready for testing.  I could not find any hubs (I recycled them all just a few weeks ago sadly) but I have a Linksys SRW208P that I can set up with port mirroring.

I am still conceptually a little behind.  I thought that the Garmin plotter would be sending GPS data and other data out the Ethernet for other plotters and devices to use.  I guess I misunderstood that.  

Mike

trudK45, 2021-01-05:
> I am still conceptually a little behind. I thought that the Garmin plotter would be sending GPS data and other data out the Ethernet for other plotters and devices to use. I guess I misunderstood that.

I am not sure your plotter does not transmit GPS data etc. but I know mine don't.
When you have everything hooked up you can check if it does.
I can think of 2 ways of doing this.
Have the radar at idle and:
1) Check in the captured Wireshark data for any string containing GPRMC.
2) Configure OpenCPN to receive data on the correct ethernet port. You will probably first have to establish what port number your plotter i sending this info on.

OpenCPN is supposed to be able to operate the radar in master mode without any plotter. This would mean that
there will be no GPS data etc. on the ethernet anyway.
If it's there we can use it with a simple setup in the OpenCPN connection menu.

/Hans

Hakansv, 2021-01-05:
Hi. I by accident found this site. I haven't though investigate it further
https://developer.garmin.com/open-source/linux/
Håkan

trudK45, 2021-01-05:
@mooselips 
I see no reason why a Garmin hub won't work.
The Garmin hub is designed to be used with a Garmin radar and the plotter should be able to power up the radar by pulling pin 5 of the ethernet cable to ground.
At this stage you should not have to worry about having a F-F connector on the path to your computer. You would just need to cut the wire to pin 5 on the patch cable so you don't mess with the input of your computer. 
You then let the plotter power up the radar scanner.
Maybe pin 5 on the computers ethernet connector is an input. In that case it wouldn't matter.
I haven't checked.
/Hans

sbalk, 2021-05-12:
Hi all,
Thanks for your work on this. I have a GMR18 (not HD) and two questions.

1. Does a GMR18 work with radar_pi? Happy to help if you need packet capture files of course.
2. Do I understand correctly that I can cut open the RJ45 cable and connect pin 5 to ground for this to work?  That would be the white with blue thread, I guess, for a US purchased version.

Stijn

trudK45, 2021-05-13:
Read this :
https://github.com/opencpn-radar-pi/radar_pi/wiki/Hardware-installation-%28Garmin%29
I would get one of these F-F connectors and do the grounding on that.
/Hans

[1 more reply]

closed 2022-04-08 by keesverruijt