opencpn-radar-pi/radar_pi #3: Garmin xHD range setting does not work for release build.

trudK45, 2017-11-08:
When building for debug and running in VS2013 the range controls does work.
It does not mater if you try to operate range with + and - in the PPI or you do it through
the menus.
I built the release as a package and installed it.
Checking with Wireshark reveals command is never sent.
To catch this you probably need to instrument the code.

canboat, 2017-11-08:
> When building for debug and running in VS2013 the range controls does work.

Does this mean that it does work if you either don’t build in debug mode and/or with other compilers and/or other platforms? Note that on Windows you cannot mix ‘debug’ versions of the plugin with ‘release’ or ‘relwithdbginfo’ versions of OpenCPN.

If you set Verbose to include Transmit (see the logfile for explanation on how to set various loglevels) do you see "transmit: turn off”?

I set Xcode breakpoints in the GarminxHD::RadarTxOff() and RadarTxOn() methods and they were called when I expected it.

If I build for debug mode with v120_xp I cannot open the plugin at all with the release version of OpenCPN.
If I build for relwithdbginfo mode it works fine and I can attach the debugger and set breakpoints in the above methods and it works fine.

trudK45, 2017-11-08:
If I compile for debug and run it in VS2013 I can change the range settings.
If I make a release package, install it under OpenCPN 4.8.0 with the opencpn.exe patch no
packets are sent to the radar.
Other commands like TxOn and TxOff are sent in both cases.
I.e. running OpenCPN 4.8.0 in the normal environment (no debug)  TxOn and TxOff are sent but not Range info.

trudK45, 2017-11-09:
I modified the code in GarminxHDControl.cpp around line 176 to

bool GarminxHDControl::SetRange(int meters) {
//  if (meters >= 200 && meters <= 48 * 1852) {
    rad_ctl_pkt_12 packet;

    packet.packet_type = 0x91e;
    packet.len1 = sizeof(packet.parm1);
    packet.parm1 = meters;
    LOG_VERBOSE(wxT("radar_pi: %s transmit: range %d meters"), m_name, meters);
    return TransmitCmd(&packet, sizeof(packet));
//  }
//  return false;
}

I.e. I didn't care what the input range value to the function was and sent it regardless.
By looking what was sent to the radar I could see that the value was huge 0x0986b7c8
which is much larger than 48 * 1852.
Where this comes from I have not yet determined. This should maybe give you Kees a hint.

Logfile from another run.
3:35:17 AM: radar_pi: Change radar range from 3516204 to 10000000
3:35:17 AM: radar_pi: Garmin xHD transmit: range 3600496 meters
3:35:17 AM: radar_pi: Garmin xHD Garmin xHD transmit 12 bytes:  1E 09 00 00 04 00 00 00 70 F0 36 00

trudK45, 2017-11-10:
Kees!

I have written out the values for range[n] for all the ranges and there seem to be something
very wrong here.
Maybe this has got something to do with the crash ?
For some reason hey do not seem to be initialized.
I may need some help here. I have a little trouble understanding the code. 
 
12:27:18 AM: radar_pi: ranges[0] = 11471417
12:27:18 AM: radar_pi: ranges[1] = 12
12:27:18 AM: radar_pi: ranges[2] = -1181843172
12:27:18 AM: radar_pi: ranges[3] = 3
12:27:18 AM: radar_pi: ranges[4] = 2027816
12:27:18 AM: radar_pi: ranges[5] = 213978936
12:27:18 AM: radar_pi: ranges[6] = 1
12:27:18 AM: radar_pi: ranges[7] = 2027760
12:27:18 AM: radar_pi: ranges[8] = 12650377
12:27:18 AM: radar_pi: ranges[9] = 0
12:27:18 AM: radar_pi: ranges[10] = 15
12:27:18 AM: radar_pi: ranges[11] = 213978936
12:27:18 AM: radar_pi: ranges[12] = 263014051
12:27:18 AM: radar_pi: ranges[13] = -715230931
12:27:18 AM: radar_pi: ranges[14] = 3516379
12:27:18 AM: radar_pi: ranges[15] = 10000000
12:27:18 AM: radar_pi: Change radar range from 10000000 to 3516379 : n=15 count=16 current_range_meters=232
12:27:18 AM: radar_pi: Garmin xHD transmit: range 15 meters
12:27:18 AM: radar_pi: Garmin xHD Garmin xHD transmit 12 bytes:  1E 09 00 00 04 00 00 00 0F 00 00 00

trudK45, 2017-11-10:
I did a git flush and git merge to get the latest fixes and now range does not work with the debugger either which is a good thing. Some thing happens when the line LOG_VERBOSE is executed.

Check Watch window: Before the line executes.
[image]

After the line is executed.
[image]

ranges are disrupted. They should not ?

canboat, 2017-11-11:
I can reproduce this, something strange happens on Windows only. macOS is fine. That should be enough for me to work out what the issue is.

trudK45, 2017-11-11:
Are you sure you allocate memory space for the vector array ? I have difficulty interpreting the code.

canboat, 2017-11-11:
Hans, if this is fixed now can you close the issue?

closed 2017-11-11 by trudK45