opencpn-radar-pi/radar_pi #4: GarminxHD : No range values in the PPI at lowest (1/8 Nm) range.

trudK45, 2017-11-08:
The header says it all.

canboat, 2017-11-08:
Do you mean that there is no value in the top left, or that there are no values on the range rings?

canboat, 2017-11-08:
1852/8 = 231,5 so let’s assume this comes in as 232, divided by 4 = 58
The current integer code checks 1852/32 = 57,875 = 57 ...

trudK45, 2017-11-08:
In the PPI there is no range in the to left for what i can see. 
This what I see at minimum range (1/8 Nm).
[image]

canboat, 2017-11-08:
Can you try again with commit 08a5758 applied?

trudK45, 2017-11-08:
Yes the 1/8 is ok now.
Other ranges are sometimes missing values. Here the inner most ring is missing value.
The number given on the next largest ring 1.12473 is difficult to comprehend.
On the garmin plotter the number of rings vary with range. In this case there are three rings
1/2 Nm between. This would be more elegant I think though it might be more work for you. 
Example 1 1/2 Nm.
[image]

canboat, 2017-11-11:
Let me know what you think now, it only draws those rings that can be represented in nice units.

trudK45, 2017-11-11:
Perfect. Only nice "round" numbers and it sets rings exactly like the Garmin Plotter.
Nice work. Only objection I have is that the outer most value is right beside the 135 degree angle value.
Maybe I am a bit fuzzy but I thing it would be a little bit easier to read if the range numbers were put at another angle. Say in between 135 and 120 degree, or 1/4 of the way up. Does this make sense ?

canboat, 2017-11-11:
Actually I'm thinking I am going to reduce the clutter by only drawing the angle every 30 degrees, like my MFD.

closed 2017-11-11 by canboat