opencpn-radar-pi/radar_pi #5: GarminxHD : Program crash.

trudK45, 2017-11-08:
Don't know if this helps.
Problemsignatur:
  Problemhändelsens namn:	APPCRASH
  Programnamn:	opencpn.exe
  Programversion:	4.8.0.0
  Programtidsstämpel:	59c69762
  Namn på felmodul:	radar_pi.dll
  Modulens version:	0.0.0.0
  Tidsstämpel för felmodul:	5a037e1d
  Undantagskod:	c0000005
  Undantagsförskjutning:	00035b8b
  OS-version:	6.1.7601.2.1.0.256.48
  Språkvariant-ID:	1053
  Ytterligare information 1:	0a9e
  Ytterligare information 2:	0a9e372d3b4ad19135b953a78882e789
  Ytterligare information 3:	0a9e
  Ytterligare information 4:	0a9e372d3b4ad19135b953a78882e789

canboat, 2017-11-08:
No :-(

canboat, 2017-11-08:
oh, gosh, it starts when the radar is already on. We had that for a while in the 3.0 development as well. Getting the timing right at startup is very difficult. Let me see if I can recollect / find what caused it last time.

canboat, 2017-11-08:
https://github.com/canboat/BR24radar_pi/issues/281 was the issue.

canboat, 2017-11-08:
@douwefokkema, this was caused by problems with the target trails code last time. Can you reproduce it in the new plugin?

canboat, 2017-11-11:
I think this should be fixed now, assuming you run with the vertex drawing method, not the shader (which is a setting on the options dialog.)

trudK45, 2017-11-11:
I am not sure what you mean by running with the vertrex drawing method. It looks like it no longer crashes when starting up with the radar transmitting. However trying to change range with the + and - in the PPI makes it crash every time. Is this waht you mean with the vertex drawing method ?

canboat, 2017-11-11:
There are two ‘drawing methods’. See Options -> Plugins -> Radar -> Preferences, then the 2nd option allows choice between ‘Shader’ and ‘Vertex’. They perform differently, on some hardware the shader is faster and on (generally older) other hardware the vertex is faster.

Until an hour ago the Shader method would crash when the length of the spokes changed. This is now fixed. The Vertex method was already OK.

trudK45, 2017-11-11:
I think I made a fetch and merge after the fix but will check again.

trudK45, 2017-11-11:
Still crashes when changing range. I am using "Vertex".

canboat, 2017-11-11:
It works for me, so it will be much easier if you are able to do some debugging.

trudK45, 2017-11-11:
I compiled for debug and ran the code in VS2013. This is what I came up with.

[image]
[image]

There is an assertion failure in msvcr120d.dll.
I am sorry to say I have no idea of how to proceed. I need some guidance here.

trudK45, 2017-11-11:
A little more info perhaps.
It mentions radar_pi.dll
reading location 0 ? Uninitialized pointer ? 

..\..\src\msw\listctrl.cpp(913): 'ListView_SetItemState' failed with error 0x00000000 (åtgärden har slutförts.).
First-chance exception at 0x0C3F2F95 (radar_pi.dll) in opencpn.exe: 0xC0000005: Access violation reading location 0x00000000.
Debug Assertion Failed!

Program: C:\Users\hasse\obeta\OpenCPN\build\Debug\opencpn.exe
File: f:\dd\vctools\crt\crtw32\stdio\output.c
Line: 1121

Expression: ("Incorrect format specifier", 0)

canboat, 2017-11-11:
I just fixed two more of those 'wrong conversion argument' lines.

If you press Retry it should open the actual debugger, and then you need to check the call stack. Somewhere up the stack there will be a LOG_XXXX message where it will use a conversion argument of the wrong size. Change the conversion argument or cast the variable.

trudK45, 2017-11-11:
Huraaay! You fixed some of the problems. Now it is much more stable, although I managed to crash it once.
So there is probably some other place as well. I will try to reproduce it.

trudK45, 2017-11-11:
Does this tell you anything ?

[image]

trudK45, 2017-11-11:
Something happens just before or at angle = 0
angle = 1439, radius = 695, m_spokes = 1440

canboat, 2017-11-11:
Your last screenshot shows a call hierarchy. That is not the call stack. This shows the call stack. It is the third dropdown in the second bar.

[image]

Here you don't see anything radar related on the call stack as I just stopped the program at some random point.

You can select the correct stack frame and then examine variables to see what causes the exception.

trudK45, 2017-11-12:
All ranges except 1 1/2 Nm worked fine.

The crash comes in PolarToCartesianLookup() when radius = 695.
It looks like you call PointInt m_max_spoke_len (733) number of times

            TrailBuffer.cpp line 101 :
                for (; radius < **m_max_spoke_len**; radius++) {
                    PointInt point = m_ri->m_polar_lookup->GetPointInt(bearing, radius);

but m_xyi  has only room for m_spoke_len (551) in this case.

Drawutil.h line 74 :

    m_xyi = (PointInt *)malloc(sizeof(PointInt) * m_spokes * **m_spoke_len**);

I think the actual  m_max_spoke_len = 705 but 733 won't hurt.

I changed "m_max_spoke_len" to "len - 1" as  in the function above and all works fine now.

I haven't committed yet. Will do that tomorrow if you don't beet me to it.

trudK45, 2017-11-12:
Now things are beginning to look good. Almost a usable radar interface.

closed 2017-11-12 by trudK45