opencpn-radar-pi/radar_pi #12: GarminxHD : FUnkar ARPA ?

trudK45, 2017-11-09:


canboat, 2017-11-09:
I rewrote large parts for this new versions, I wouldn't be surprised it's broken right now. ARPA is Douwe's speciality, I hope he has time to look into this.

douwefokkema, 2017-11-10:
I tried the ARPA on the Scan_1.pcap file. Replayed with fixed position and heading. I noticed speeds in a (seemingly) anchored target. But that target moved up and down over 120 meters on my overlay. This causes the target to pick up some speed in various directions. There are also some targets that show and hide all the time. This may cause an ARPA target not to correspond to a radar blob. However, when a target is not seen in the last sweep of the radar, the ARPA target should turn yellow.
I will look into this further. A handicap in checking this is that I don't have heading info that runs simultaneous with the recording (with the Navico radar I use the heading on radar).
Hans, which recorded file is best to check ARPA?

canboat, 2017-11-10:
Douwe,

I was wondering that maybe the different # of spokes is not accounted for correctly in the ARPA code. But let’s start with checking that ARPA still works with Navico after my rewrite for the variable radar # and spoke #.

trudK45, 2017-11-10:
I can prepare a new .pcap and NEMA data to along with it where we do have targets.
Would this help ?
I will be out during the day but can have this ready in the evening.

douwefokkema, 2017-11-10:
Yes, certainly worth a trial. When you output the NMEA data as UDP on the net it should be logged synchronously in the pcap file. Just take a port number (eg 5555)

canboat, 2017-11-10:
Would you please record the NMEA (NEMA is something else) data as an UDP stream in the pcap file? That way they are synchronised in time.

You can use OpenCPN for this. Add an output UDP port 5001, IP address your broadcast IP address. for instance if your radar ethernet card has IP address 172.16.2.40 set the IP address to 172.16.2.255.

Then make sure you send all NMEA sentences (Output filtering -> click “drop sentences” and add no sentences.)

trudK45, 2017-11-10:
I have uploaded a zipped .pcap file (ARPA_alarm.zip) with ARPA on, Alarm on. 30 sec long.
Broadcast address for NMEA data=176.16.15.255
NMEA includes : GPRMC, GPGBS, HCHDG (10 Hz) and AIS-data.

canboat, 2017-11-21:
Oops, reopening as I used the wrong #12 instead of #26 in commit a59115e.

douwefokkema, 2017-11-24:
Yes, ARPA is functioning as it should, at least in 58a58c12

commit a59115ef (2017-11-21): Fix #12. Only show manual value in auto mode for Garmin.
closed 2017-11-24 by douwefokkema