opencpn-radar-pi/radar_pi #6: Rain and Sea clutter can't be set to 0.

trudK45, 2017-11-09:
When setting these to 0 they change to 1 automatically after 1 second or so.
This is only something wrong in the displayed value.
Locking on the Garmin plotter these are off.

canboat, 2017-11-09:
Like gain, I had to make some assumptions on what the Garmin sends back when the value is off (I'm using '0' instead of a separate OFF value. Maybe that is too simplistic.)

Can you upload some pcap files with:
- sea clutter 40
- sea clutter 1
- sea clutter 0 (is this even possible?)
- sea clutter off

Am I right in assuming that the maximum value is 100?

trudK45, 2017-11-09:
Yes 100 is th maximum value.
I have uploaded 4 file to my dropbox.
Names are self explanatory.

canboat, 2017-11-11:
Hans, is this ok now?

trudK45, 2017-11-11:
Can't test this at the moment, as the plugin crashes before I have the chance to go into that menu.  Will check when possible.

trudK45, 2017-11-12:
Sea clutter seems ok. Rain clutter does not.

canboat, 2017-11-12:
OK, lots more work done on Off/Manual/Auto. Please report any remaining issues.

trudK45, 2017-11-12:
I checked the behavior of Sea and Rain clutter menus. Both can be set to 0.
However all the menus now flicker with an extra button underneath. I can make a recording if you need.
There may be an issue with auto but we should get it stable first.

trudK45, 2017-11-13:
Uploaded a recording to my dropbox.
[https://www.dropbox.com/preview/RecordingGarmin/ice_video_20171112-174324.webm?role=personal](url)

canboat, 2017-11-13:
> I checked the behavior of Sea and Rain clutter menus. Both can be set to 0.

And Off, I hope?
> However all the menus now flicker with an extra button underneath. I can make a recording if you need.
> There may be an issue with auto but we should get it stable first
> 
> 

Yes, I added an extra call to EnterEditMode() which turns out is called way more often than I thought, but it doesn’t flicker for me. The extra call was to make sure that if you change a value on a plotter or 2nd OpenCPN device that the value is updated even if you are editing it.

You can take out the extra call to EnterEditMode in ControlsDialog.cpp line 1720 and see whether that fixes any flicker.

trudK45, 2017-11-13:
If I take out the line you mention there is no flicker anymore.
Off almost work. If you activate it (push off) the old percentage still is shown. 
You have to leave the menu and then return to see the off word off.

canboat, 2017-11-14:
> There are still issues with the flicker. I did not push the change at line 1720 in ControlsDialog.cpp
> so that is probably back in action and the reason why it flickers.

The EnterEditMode is only called now when the value actually changes, so this is weird it should not be called that often. It is frustrating to try to debug without a live radar, can you check whether the EnterEditMode is called (too) often using your live system? Attach the debugger or start O from VisualStudio, add a breakpoint in RadarControls.cpp where it is calling EnterEditMode, I think the line number changed a little.

> In the menu it's not possible to go from Off to Auto. I think one should be able to.

You're not specific, but I guess the Auto button is enabled but if you click it doesn't work? Can you check that it sends commands when you do this? (either verify the log records or attach the debugger + breakpoint in GarminxHDControl.cpp line 306 (case CT_SEA: etc)) maybe the commands are incorrect?

trudK45, 2017-11-16:
I had a look in ControlsDialog.cpp and I can't understand the code. Please enlighten me.
At line 219 we have this :
bool RadarControlButton::ToggleState() {
  RadarControlState state = m_item->GetState();

  if (state == RCS_OFF) {
    state = RCS_MANUAL;
    return true;
  } else if (state == RCS_MANUAL && m_autoValues == 0) {
    state = RCS_OFF;
    return false;
  } else if (state < RCS_MANUAL + m_autoValues) {
    state = (RadarControlState) (state + 1);
  } else {
    state = RCS_AUTO_1;
  }
  SetState(state);
  return m_autoValues > 1;
}

state is  = RCS_OFF , then pushing Auto will set state to Manual. But then later
line 1473 is executed and sets state = RCS_OFF

line 1473 : m_timed_idle_button->SetState(RCS_OFF);

Is this the way it should be ?

canboat, 2017-11-18:
You are right it is wrong. Note that that function is only used when you press the Auto button. It has been reworked, in addition to some refactoring/simplification of the entire control stuff. More buttons are now standard RadarControlButtons with no override of their labeling code.

It seems to work a lot better now and I hope that the 'flickering' (which, after Douwe explained it to me, was that two sizers were shown simultaneously, where only one is supposed to be shown.)

trudK45, 2017-11-18:
The Sea clutter menu now seem to work fine without any flicker, but, you have to be real quick to activate Transmit after you pushed "Power Status Standby" button. The next menu is only shown 1/2 second or so and then returns to the previous menu (main). Don't have time right now to do debugging.

trudK45, 2017-11-19:
I have uploaded a recording ice_video_20171119-004558.webm.
I have set a break point at line 1510 in ControlsDialog.cpp where resize is set to true.
The menu is then reset to start menu at line 1786 in the Fit() function.
Fit() is a wxWindows routine.
I guess there are some state/parameters that are wrong.

canboat, 2017-11-19:
It wasn't really 'wrong' it was an attempt to 'help' the user when the power is off not to show any submenu anymore. It should have excluded the power menu. I have now removed the automatic menu change all together.

canboat, 2017-11-24:
Hans, can this be closed now?

closed 2017-11-24 by trudK45