opencpn-radar-pi/radar_pi #7: GarminxHD : Gain control shows only Auto High.

trudK45, 2017-11-09:
When changing value the new value in % is shown a short wile and then changes to Auto High.
The button Auto seem to have no effect.

canboat, 2017-11-09:
The way the code works is that it assumes that the radar listens and adjusts the on-screen value and sends a request to the radar. Either the radar is not listening or I misinterpreted how the garmin xHD sends the 0x924, 0x925 and 0x91d packets. 

Could it be that it sends 0x91d even when auto is off?

Can you either fix this yourself in the code or upload 3 pcaps of about 10s, with:
- gain at some value 
- auto gain low
- auto gain high

trudK45, 2017-11-10:
Uploaded 3 files to my dropbox.

commit 644bd5dd (2017-11-11): Fix #7. Remember auto gain state to be used in next messages
closed 2017-11-11 by canboat