opencpn-radar-pi/radar_pi #14: GarminxHD : Timed Transmit menu does not work correctly.

trudK45, 2017-11-10:
Timed Transmit says "5 min" and when you push + nothing happens. Pushing - setts Timed Transmit to off.
 To get it back to "5 min" you have to go into Runtime and change value.
The push buttons ar named + and - but change the runtime value bu 10 sec.
Either let the runtime change by 1 sec. or label the pushbuttons +10 and -10.
I would prefer the later.

trudK45, 2017-11-20:
I had a look at the code in GarminxHDCcontrol.cpp No timed transmit commands are in there.
Is Timed transmit implemented in the radar not used, but handled by code ?

canboat, 2017-11-20:
> Is Timed transmit implemented in the radar not used, but handled by code ?

Correct. The plugin does it itself using no support from the radar.

See issue #25.

trudK45, 2017-11-20:
Which one is it ? issue #25  says to use Garmin functionality, this say not to.

canboat, 2017-11-20:
Issue #25 is a lot of work so maybe we do this one first.