opencpn-radar-pi/radar_pi #18: True trails flip to opposite side

douwefokkema, 2017-11-11:
After trailing for a while, the true trails flip to the opposite side of the image. On the image: ship moving N at 5 kn, so do all the targets.

[image]

canboat, 2017-11-11:
Which pcap file is this? And are you going to fix this or do you suggest that I take a look?

douwefokkema, 2017-11-11:
This is Scan_1.pcap. It is a stationary file played with a moving GPS. I will take a look tomorrow.

douwefokkema, 2017-11-20:
I fixed this in commit 160af32ed. Later it did not work anymore, I fixed the trails again in a0de59148. Today I verified trails again, functional in latest commit, 8e7ed30a. All tested with Scan1.pcap. But zooming of trails I have not tested and might be in error still. For this I need a replay (pcap) file that contains zooming over various ranges. Is there such a file? I do have access to Hans his dropbox, there I have the Scan1 from.

douwefokkema, 2017-11-22:
I have been testing now the zooming of trails. After correcting the zoom_factor to its inverse it seemed to function, but some trails were disappearing mysteriously when zooming. Real hard to find as it was only some of the one million pixels now and then. Issue was that the calculation of indices was modified to type size_t. When calculating with size_t as example 10-20+20 is not the same as 10+20-20 as it is unsigned. So I changed the indices to int. All fine again, tested with the Send_NMEA_COM GPS simulation tool (from Håkan, very good, allows you to adjust speed and heading). Requires further testing i the field. See commit fdf79794a

Hakansv, 2017-11-22:
After that commit the trails are much better and kept after zoom in and out. Emulator test only so far.

Hakansv, 2017-11-22:
Douwe.
A short emulator timed transmit test: After timed pause and the transmit begin the first sweep draw old trails probably some memory left over? They are quickly erased and the fresh ones are drawn. We may delete old trails memory when radar transmit is not active?

trudK45, 2017-11-22:
I think we can close issue #21
The gray areas are no longer there.

douwefokkema, 2017-11-24:
I have been testing now the zooming of trails. After correcting the zoom_factor to its inverse it seemed to function, but some trails were disappearing mysteriously when zooming. Real hard to find as it was only some of the one million pixels now and then. Issue was that the calculation of indices was modified to type size_t. When calculating with size_t as example 10-20+20 is not the same as 10+20-20 as it is unsigned. So I changed the indices to int. All fine again, tested with the Send_NMEA_COM GPS simulation tool (from Håkan, very good, allows you to adjust speed and heading). See commit fdf79794a. Has been tested by Håkan as well now.

closed 2017-11-24 by douwefokkema