opencpn-radar-pi/radar_pi #19: GarminxHD : Gain value takes long time to come up with correct value in the menu.

trudK45, 2017-11-11:
Radar is powered on. It has a gain value of something. lets say 40%.
Starting up OpenCPN,  Commanding transmit,  Go into the menu for setting gain.
There it shows 0% Gain. After a while it might present the correct value 40 % but not always.
I have been waiting over 1 minute and nothing has happend.
I you wait in Standby for a while the value might be correct.
It you change the value with + the value is sent to radar setting it to 1%.
It looks like the gain status is not always read and loaded into the present value at init.

canboat, 2017-11-11:
Is this with the latest changes?

Try running with VerboseLog=1 then check the logfile for gain + Gain, and please update this logfile.

trudK45, 2017-11-11:
Uploaded opencpn_gain.log to my dropbox. From start gain on radar is 50%. Entering menu shows gain =0%. Pushing + five times, starts at line  2341  ends in line 2497.
Line 288 says gain 5000 which corresponds to the 50% i guess.

canboat, 2017-11-11:
I will check with my Navico radar tomorrow.

canboat, 2017-11-11:
In the logfile I see
```
8:30:41 PM: radar_pi: Radar A ControlDialog: Adjusting Gain by 1 from 0 to 1
8:30:41 PM: radar_pi: Garmin xHD Gain: 1 auto 0
8:30:41 PM: radar_pi: Radar A ControlDialog: Set Gain value 0 -> 1, range=0..100
8:30:41 PM: 0x0924: autogain 0
8:30:41 PM: 0x0925: gain 100
8:30:42 PM: 0x0924: autogain 0
8:30:42 PM: 0x0925: gain 100
8:30:42 PM: 0x091d: auto-gain mode 1
8:30:42 PM: radar_pi: Radar A ControlDialog: Set Gain to auto value 0, max=2
8:30:42 PM: radar_pi: Radar A ControlDialog: Adjusting Gain by 1 from 1 to 2
8:30:42 PM: radar_pi: Garmin xHD Gain: 2 auto 0
8:30:42 PM: radar_pi: Radar A ControlDialog: Set Gain value 1 -> 2, range=0..100
8:30:42 PM: 0x0924: autogain 0
8:30:42 PM: 0x0925: gain 200
8:30:42 PM: radar_pi: Radar A ControlDialog: Adjusting Gain by 1 from 2 to 3
8:30:42 PM: radar_pi: Garmin xHD Gain: 3 auto 0
8:30:42 PM: radar_pi: Radar A ControlDialog: Set Gain value 2 -> 3, range=0..100
8:30:42 PM: 0x0924: autogain 0
8:30:42 PM: 0x0925: gain 300
8:30:43 PM: radar_pi: Radar A ControlDialog: Adjusting Gain by 1 from 3 to 4
8:30:43 PM: radar_pi: Garmin xHD Gain: 4 auto 0
8:30:43 PM: radar_pi: Radar A ControlDialog: Set Gain value 3 -> 4, range=0..100
8:30:43 PM: 0x0924: autogain 0
8:30:43 PM: 0x0925: gain 400
8:30:43 PM: radar_pi: Radar A ControlDialog: Adjusting Gain by 1 from 4 to 5
8:30:43 PM: radar_pi: Garmin xHD Gain: 5 auto 0
8:30:43 PM: radar_pi: Radar A ControlDialog: Set Gain value 4 -> 5, range=0..100
8:30:43 PM: 0x0924: autogain 0
8:30:43 PM: 0x0925: gain 500
8:30:44 PM: 0x0924: autogain 0
8:30:44 PM: 0x0925: gain 500
8:30:44 PM: 0x091d: auto-gain mode 1
```

I see you pressed + five times, and I see the radar reacting with the 0x924 0x925 packets, with the values I would expect for gain value 1 (100), 2 (200) etc... Are you saying the UI doesn't show 1 - 2 - 3 - 4 - 5 as you press + ?

trudK45, 2017-11-11:
No. What I am saying is that the radar_pi overrides the set value of, in this case 50% and always sets it to 0% when you fire up OpenCPN. You should not have to increase the gain from zero every time you start up OpenCPN. This does not happen with other settings what I can see. I.e Rain and Sea clutter.
The program should detect what is already the gain value that the radar is reporting over the ethernet and start from there. Not from 0.

trudK45, 2017-11-11:
If you look further up the file it says :  0x0925: gain 5000
This means it has detected that the radar is set to 50% gain before you send any gain set command.

canboat, 2017-11-11:
aaahh, ok now I understand what you are saying. OK. Now I can reproduce by playing various PCAP files after each other.

canboat, 2017-11-11:
I think I fixed it now. Can you test again?

trudK45, 2017-11-11:
Sorry but now the program crashes when starting transmit despite the fact the radar is in idle to start with.
It manages about 200 degrees then I get the crash. I think we have to catch this first. I was just on my way debugging when interrupted to check the Gain. The previous state of the program was quite stable and I had to provoke it to get it to crash.

trudK45, 2017-11-12:
Yes, this is ok now.

closed 2017-11-12 by trudK45