opencpn-radar-pi/radar_pi #20: GarminxHD : Circle at outer boarder on certain ranges.

trudK45, 2017-11-12:
After the change "Refactor variable spoke length" we now get rings at the outermost limit of the range.
Probably due to presenting data to far out which then is static and not updated with radar data.

[image]

canboat, 2017-11-12:
Reproduced with Sea_Clutter_0.pcap, both Vertex and Shader, even with extreme range ring off.

canboat, 2017-11-12:
interestingly, that is from a range with spoke_len = 733 == spoke_len_max...

trudK45, 2017-11-12:
Yes but the 733 includes the header with data describing the radar data. The spoke is only 705 long i.e 28 bytes less. These 28 bytes are the "header". The length of the spoke s specified in scan_lenght_bytes_s as it is called in my document.

commit 3789704b (2017-11-12): Fix #20. Use better version of radar_line header.
closed 2017-11-12 by canboat