opencpn-radar-pi/radar_pi #21: GarminxHD : Gray areas when increasing range.

trudK45, 2017-11-12:
This refers to the PPI only. It looks like the plot memory is reset when you go to a lower range but not when you go to a higher. I have uploaded a recording, ice_video_20171112-174324.webm  to my dropbox.

canboat, 2017-11-13:
The intended behaviour is that the target trails are zoomed in and out when you change range, e.g. in both directions.

canboat, 2017-11-13:
Hans, can you share your dropbox with Douwe so he can take a look?

trudK45, 2017-11-13:
I  think I already have done that. Did you try Douwe ?

canboat, 2017-11-18:
I get weird results with trails as well, on Windows they seem to appear at strange times and on macOS they don't disappear if I set them to a low # of seconds.

douwefokkema, 2017-11-20:
I have access to your dropbox, using Scan1 most of the time. Is there a pcap where I can see this issue (pcap with zooming on it) so I can replay this issue. I do not need a video.

trudK45, 2017-11-20:
Jobs done.
I have uploaded a .pcap to my dropbox., Gray_areas.zip

douwefokkema, 2017-11-21:
Thanks, I will work with it right now.

douwefokkema, 2017-11-22:
This issue is identical to #18. See there for further reactions.

douwefokkema, 2017-11-23:
Hans reports that grey areas are no longer there. Close this issue. Further remarks on trails under #18

closed 2017-11-23 by douwefokkema