opencpn-radar-pi/radar_pi #22: Visible gain would be nice.

trudK45, 2017-11-13:
It would be nice if we could have the gain value displayed somewhere in the PPI.
Also, I would be happy to have a "warning" displayed if Rain or Sea clutter are turned on.
It can be dangerous if you are unaware that you have these on and ships are attenuated below level of detection. This may be later issue, enhancement.

canboat, 2017-11-13:
Good idea. We can copy how the radar screens on the plotters do this.

trudK45, 2017-11-13:
The plotter has the gain visible but no warning about sea and rain clutter. That was my own idea.

canboat, 2017-11-14:
Added these to the bottom right of the PPI window.

In addition to showing the value and a graph (arc), these are color coordinated:
Off = Grey
Manual = Reddish
Auto = Greenish

We can fiddle with the colors or even make them selectable.

trudK45, 2017-11-14:
The display looks wonderful. Really like it. 
In the menu it's not possible to go from Off to Auto. I think one should be able to.
There are still issues with the flicker. I did not push the change at line 1720 in ControlsDialog.cpp
so that is probably back in action and the reason why it flickers.

commit a8640e9c (2017-11-23): Fix #22. Make this even better.
Show Auto name, if on auto. Don't show value if on Auto + not Garmin.
closed 2017-11-19 by canboat