opencpn-radar-pi/radar_pi #24: GarminxHD : Interference Rejection menu should be simpler.

trudK45, 2017-11-19:
The Interference Rejection can only be on or off.
+ and - can be take away.
Button should change Off/On depending on state.
This function is called Cross Talk Rejection on the Garmin plotter.

canboat, 2017-11-20:
At the moment we can only do an off/on using + and -, not an On and Off button. I realise this is sub optimal.

trudK45, 2017-11-20:
To operate the function you need to use plus and off. Minus has no effect at all, besides chaining the test to off temporarily. The state of the radar does not change.
If you fiddle with off/on on the plotter you can have minus and plus working instead and not plus. But then off becomes on and minus is off.

canboat, 2017-11-20:
OK that means there is still something wrong with the mapping of 0 and off. I’m sorry this is much more messy than I thought and it is hard to test without a live radar.

canboat, 2017-11-20:
Strange, on macOS I just get the '+' and '-' buttons, not the Off button (which is what I expected) but on Windows there is this extra Off button.

canboat, 2017-11-20:
OK just committed change that removes the "Off" button, leaving the + - buttons. If clicking Off does not work can you check yourself that the code sends the right commands to the radar?

See this section in `garminxhd/GarminxHDControl.cpp`:

    case CT_INTERFERENCE_REJECTION: {
      LOG_VERBOSE(wxT("radar_pi: %s Interference Rejection / Crosstalk: %d"), m_name.c_str(), value);
      pck_9.packet_type = 0x91b;
      pck_9.parm1 = value;

      r = TransmitCmd(&pck_9, sizeof(pck_9));
      break;
    }

closed 2017-11-24 by canboat