opencpn-radar-pi/radar_pi #25: Timed transmit should use Garmin hardware feature

canboat, 2017-11-20:
When timed transmit is used it should use the Garmin xHD builtin functionality.

It should send the appropriate values. What I don't really understand yet is what we should send at which time. The spreadsheet says there are two messages 0x943 SetStandbyMinutes and 0x944 SetTransmitMinutes.

Do we send SetStandbyMinutes "x" once when we want the radar to stop in x minutes and then once it standby do we sent SetTransmitMinutes "y" once and then it it will start after "y" minutes? Does the radar give feedback on this?

trudK45, 2017-11-20:
Timed transmit has 3 commands.
1) Timed transmit on/off             0x942
2) Timed standby                        0x943
3) Timed transmit                        0x944

When you have set these the radar does the loop transmit -> standby -> transmit and so on by itself.
Stop it by sending Timed transmit off.
I am not sure how it is reported. I will investigate this.
Leave it to me. Which you have already done. :0)

canboat, 2018-01-15:
@trudK45  does this work now?

trudK45, 2018-01-15:
No, this is not done yet. We have been to busy getting the boat ready for launch and then moving around. I have focused on trying to understand the code.