opencpn-radar-pi/radar_pi #26: Navico: Auto Gain and Sea Clutter info

Hakansv, 2017-11-20:
Auto gain and Auto sea clutter status is showing last manual value only. If there are no response from radar to view we better hide the values when in Auto.

canboat, 2017-11-20:
It would be interesting to see if the Garmin radar tells the plugin what value it is using for Auto. This is what I am hoping, really. If it does then it would be cool to show them, but then we need a way for the radar-type specific code to tell the generic code how to handle the value when Auto is active.

canboat, 2017-11-20:
@trudK45  can you comment on the above?

trudK45, 2017-11-20:
Yes, I can confirm that the Garmin reports gain and auto mode during Auto gain.
Sorry for not answering earlier but the internet was down.

Hakansv, 2017-11-21:
An observation:
If I set for example Sea Clutter to static 0% and then to Auto the control value shows "Auto". If the static value is >0% e.g. 10% and then to Auto the control shows "Auto 10%". The same appears for Gain and others.

canboat, 2017-11-21:
Fixed in commit a59115e. This is now settable per radar type using a boolean in RadarInfo.

closed 2017-11-21 by canboat