opencpn-radar-pi/radar_pi #27: Navico: Side lob suppression. Miss Auto option

Hakansv, 2017-11-20:
Side lob suppression: Auto option is missing. When in Auto the status is though viewed correct. (Achieved after set to Auto using BR24radar_pi)

canboat, 2017-11-20:
Oh can you try changing this:

HAVE_CONTROL(CT_SIDE_LOBE_SUPPRESSION, CTD_AUTO_NO, CTD_DEF_ZERO, CTD_MIN_ZERO, CTD_MAX_100, CTD_STEP_1, CTD_PERCENTAGE)

change CTD_AUTO_NO to CTD_AUTO_YES

canboat, 2017-11-20:
In navico/NavicoControlSet.h

> On 20 Nov 2017, at 13:17, Kees Verruijt <kees@verruijt.net> wrote:
> 
> Oh can you try changing this:
> 
> HAVE_CONTROL(CT_SIDE_LOBE_SUPPRESSION, CTD_AUTO_NO, CTD_DEF_ZERO, CTD_MIN_ZERO, CTD_MAX_100, CTD_STEP_1, CTD_PERCENTAGE)
> 
> change CTD_AUTO_NO to CTD_AUTO_YES

Hakansv, 2017-11-20:
Yes, I can test and come back after live test. (Now home and have to go back to the boat for live test)

Hakansv, 2017-11-21:
Tested OK. I couldn't make a direct push. A PR is waiting instead.

closed 2017-11-21 by Hakansv