opencpn-radar-pi/radar_pi #28: Navico: Noise rejection is allways "Off"

Hakansv, 2017-11-20:
Noise rejection has Off-Low-High. Neither Low nor High is working. When activated there’s a short glint of Low/High but it’s reset to “Off” within a second.
Note: It may fixed be commit "Issue #24. Remove 'Off' button from controls that only have values." I'll check.

Hakansv, 2017-11-22:
Not fixed by commit "Issue #24"
I found no reason for the failure. When it's set to 1 the report back is always 0 = Off. No typos compared to B24radar_pi. 
Let's see if your eyes or head is better.  If not solved before I make a pcap next time in boat. 
I'm not 100% sure this was working on a BR24 before. Correct register? I'll compare to BR24radar_pi function as well.

Hakansv, 2017-11-23:
Now again tested live on BR24:
Issue still remains. If I set noise rejection to 1=low the radar response seems misinterpreted and switch the control value back to 0=Off. This statement is based on the fact that if I instead use BR24radar_pi the actual radar response is 1=Low. And using BR24radar_pi the state is able to change back and forth between all three values. Leaving BR24radar_pi at noise rejection "Low", radar_pi control status shows 0=Off.

canboat, 2017-11-23:
This is turning a bit into a mystery, as it works perfectly for me with my 4G (live, not playback) -- if I change something in the control I see it reflected on the MFD and vice-versa.

canboat, 2017-11-23:
Can you upload a logfile with loglevel receive so we can examine how the plugin interprets RadarReport_08C4_18?

Hakansv, 2017-11-23:
OK - will do next time aboard. Hopefully tomorrow.

trudK45, 2017-11-23:
On the Garmin this behaves "funny".
When you operate the function on and off the info on the menu follows correctly.
I.e. it detects the status of the radar and report it.
If you use radar_pi to activate and deactivate the Noise Rejection it does not work right.
If Off and you click plus the Noise Reject goes to On and for about a second this is shown
in the menu. Then the text goes to Off again. Clicking + repeats the status shown (On for 1 sec. then Off).
Clicking minus have no effect. The radar Noise Rejection still is on. This might be the same as on the Navico. Difference is I run live ?

canboat, 2017-11-23:
It works for me with 4G. Either you find out what is wrong yourself or you give me more to work on.

That means a log file showing both the commands sent, the radar reports and values coming from the radar. If you can’t interpret the logfiles so won’t I. 

Note: the behaviour described by trudK45 is what I would expect if the radar is not accepting the commands. The immediate update in the control is what the plugin *thinks* it is telling the radar. Either it is sending the wrong command, or no command, or the radar report is interpreted incorrectly.

Try matching up the binary log of the radar report with the fields that we think are this particular field to see if we interpret the report correctly.

trudK45, 2017-11-24:
Fixed the Garmin code. Now This works for the Garmin.

canboat, 2017-11-24:
Cool, so it turned out to be the code didn't send the right commands to the xHD radar.

trudK45, 2017-11-24:
The command sent was correct but looking at what the plotter sent there where in total 3 packets. I had anticipated 1 was enough. I may later add packets to the other command for the Garmin code to fully emulate the Garmin plotter. Up till now I have only sent the commands I saw was needed.
Also, the value of the length of the packages where wrong.

Hakansv, 2017-11-24:
On BR24:
Nothing wrong with radar_pi at all. Instead radar_pi code compared to BR24radar_pi is excellent! 
Using radar_pi it's obvious this function is not available on the BR24. At least not in register 0x21, 0xc1. This command yields no response from BR24 to 18 - 08 C4. Since BR24radar_pi don't show true response, only what's sent, this was not detected before.
What I could do is to send various not used commands and check for any result? Any suggestion for plausible commands to test?
See also some OCPN logs and pcap's from today's test: https://www.dropbox.com/s/yb51d0i436rurlc/Radar_pi-BR24_NoiseRed.zip?dl=1

canboat, 2017-11-24:
ahhh!

That actually makes sense because I know that when I upgraded in 2013 from BR24 to 4G I gained a lot of extra buttons on my plotter. The problem is I forgot to make screenshots of the plotter controls with the BR24 so I cannot recollect what exactly was added, and then we started adding all those 4G commands to the plugin…

So my suggestion is that we disable this button for BR24. Can you try this (edit the controlset, make an extra #ifdef around this one so that it is not used for BR24…)

I don’t think you will find any ‘hidden’ commands.

Hakansv, 2017-11-24:
OK-Fine. I'll make PR for this and the sea state change. (I'll close the old one)

closed 2017-11-24 by Hakansv