opencpn-radar-pi/radar_pi #30: Radar window showing a wrong heading

douwefokkema, 2017-11-20:
In commit 8e7ed30a125f5 the head up radar window is showing a wrong heading for the 4G radar. see picture.
[image]

In this picture trails are green, radar is single color.
Heading is 0, radar window showing 240. Overlay is OK.

canboat, 2017-11-20:
Did you select 'disable heading on radar'? Otherwise the PPI window will prefer the radar heading, if present. This means that in emulation circumstances it is easy to get a 'strange' result...

If I use a pcap from Merrimac where I am at a known heading of 150 degrees I get 150 degrees on the PPI, so that matches.

douwefokkema, 2017-11-21:
You are right, after all this Garmin I forgot about heading on radar on Navico (better indeed)

closed 2017-11-21 by douwefokkema