opencpn-radar-pi/radar_pi #33: Navico BR24: Can't switch to Range Auto

Hakansv, 2017-11-21:
At program start I've Range: Auto.
I can change to a static range e.g. 1 NM but if I use the Auto button after that the status is not changed to Auto but the static range remains. 
Upon next program start the Auto is back.
(I'm in the boat now and can test any change.)

Hakansv, 2017-11-21:
A reported range is also not correct, neither in Auto nor manual. At reported 1 NM the actual distance to the extreme range ring is 1.77. That factor is consistent for all ranges.  (An AIS target range is on a correct distance though. So nothing wrong on my screen)

canboat, 2017-11-21:
This is a very brief report. 

It is unclear to me what the problem is exactly, as there is always a discrepancy between the commanded range and the actual distance that the radar spokes have. In the past the code knew what these were, but now it just asks for a value and just looks at what the radar does when it produces spokes. In other words, it asks the radar for 1 NM, then listens for a report (this will give ‘1 NM’ otherwise the control would tell you something else) and then looks in the spokes for the actual range.

In the past we would receive 2693 m when commanded to do 1852 m, for a 4G. Not sure about your BR24. A constant 1.77 sounds wrong though, these ‘overscans’ are generally different for every range.

So can you tell us:

1. Is the scale of the overlay correct?
2. Are AIS symbols on the PPI at the right scale (distance from center?)
3. Are radar targets on the PPI at the right scale?

Ohhh, one thing — maybe the code is using the wrong 4G header to determine the actual spoke range?

Hakansv, 2017-11-21:
1. All targets and land contours are correct sized and positioned.
2. Yes
3. Yes
So I'd say what we draw on chart and PPI is correct. The extreme range rind and the value reported back from radar is not quit correct.

canboat, 2017-11-21:
OK, then I still think it works as designed. There is a substantial ‘overscan’ but that is how the radar desires to work.

Can you get all the way to 24 NM and what is the overscan % and m then?

Hakansv, 2017-11-21:
It's still factor 1.77.  Ring at 42.5 N and radar says 24 N

Hakansv, 2017-11-21:
More on Auto function:
If I switch to a static range and try to got back to Auto i don't work.
But if I change language, e.g. English to Swedish or vice versa, the menu is redrawn and range is switched to Auto! 
Is this all about control update and not what's actually coming from the radar?

canboat, 2017-11-21:
A radar doesn’t have auto range — that is just an invention of the plugin.

Hakansv, 2017-11-21:
Of course! .... Blackout.
So - control update only in other words.

canboat, 2017-11-21:
I looked into this briefly and there is still some remnants of the old plugin that need to be removed/moved to the new way of doing things; I refactored RadarControls but forgot to do all of RadarRangeControl. It still can't be made a normal RadarControl as the list of values is dependent on the units in use.

trudK45, 2017-11-22:
I am not sure what the Auto range function does but I suspect that it should match the size of the chart.
I his is the case then it does not work for Garmin either.

canboat, 2017-11-23:
OK, it works for me now. Note that it only works if you have the overlay on. In fact, if you don't have chart overlay enabled it is now no longer possible to select Auto.

canboat, 2017-11-23:
> At reported 1 NM the actual distance to the extreme range ring is 1.77.
I also verified that the Navico radar does indeed have a large overscan, but this is as designed. At all ranges did the display of the range match up with the plotter, so this is just how it is.

Hakansv, 2017-11-24:
Now it works very good on BR24 as well. Thanks

closed 2017-11-24 by Hakansv