opencpn-radar-pi/radar_pi #36: BR24/Emulator: Large movement trails reset > False calculation?

Hakansv, 2017-11-22:
Trails sometimes disappear due to MARGIN(=100) overleap. Are the calculations in Trailbuffer.cpp row ~300 correct? It's not checked by debug so I can't say what's happening. Pls have a look.
Log file examples:
1:08:50 PM: radar_pi: Navico BR24 Large movement trails reset

9:14:09 AM: radar_pi: offset lat too large 684
9:14:09 AM: radar_pi: Emulator Large movement trails reset
9:14:09 AM: radar_pi: Emulator Large movement trails reset
9:14:11 AM: radar_pi: Emulator Large movement trails reset
9:14:12 AM: radar_pi: Emulator Large movement trails reset
9:14:15 AM: radar_pi: Emulator Large movement trails reset
9:14:16 AM: radar_pi: Emulator Large movement trails reset
9:14:17 AM: radar_pi: Emulator Large movement trails reset
9:14:19 AM: radar_pi: Emulator Large movement trails reset
9:14:22 AM: radar_pi: Emulator Large movement trails reset

9:56:49 AM: radar_pi: offset lon too large 3342386
9:56:49 AM: radar_pi: offset lat too large 2883662
9:56:49 AM: radar_pi: Emulator Large movement trails reset

douwefokkema, 2017-11-22:
True trails require a reliable GPS position and a heading. I do not know how the emulator handles this, I never use the emulator.

Hakansv, 2017-11-22:
Both, pos/HDT, are available and trails are normally fine. But I also got this log message from BR24 live test. 
This issue was more a hint to check the calculations since a lot has changed and I never saw these log messages before. If all seems OK we better close this and I'll check during real boat movements next spring.

Hakansv, 2017-11-23:
From today's BR24 test:
Below log entries appeared after a short timed transmit test. Could it be that old trails storage should be deleted if radar transmit is switched off?
09:41:40: radar_pi: Navico BR24 transmit: turn Off
09:41:47: radar_pi: Navico BR24 transmit: turn on
09:41:48: radar_pi: Navico BR24 data lost
09:41:49: radar_pi: offset lon too large 909651507
09:41:49: radar_pi: offset lat too large 875639088
09:41:49: radar_pi: Navico BR24 Large movement trails reset

douwefokkema, 2017-12-06:
This was indeed a real issue. Thanks for noticing it! Solved in 82dc7fcf424

closed 2017-12-06 by douwefokkema