opencpn-radar-pi/radar_pi #38: After changing radar type trails duration shows "Off"

douwefokkema, 2017-11-24:
After changing radar type (after a "Reset radar types) trails duration shows "Off", whiles trails are on. See attached picture (multiple different radars working fine!)
[image]

canboat, 2017-11-24:
interesting as the code should destroy + recreate all objects such as canvases and dialogs.

canboat, 2017-12-02:
I don't understand this after checking myself. If you reset the radar types some settings revert to the initial default. What is the problem with that? I don't think a real user will ever reset the radar types (only when buying a new one...)

douwefokkema, 2017-12-03:
The problem is that trails duration doesn't have an "Off" (if its right). Trails are switched off with the "Trails motion" button. In normal operation there is indeed no "Off" on trails duration. But after changing radar types there is an "Off" shown. See picture. his is not a serious error, but I still thought I should mention it.
A user will not REset radar types often. But I like to enable or disable a Navico B without destroying the Navico A.

commit 62190199 (2017-12-08): Fix #38. Fiddle with loading/saving in the right order.

The problem is that we store settings in the RadarInfo object which makes that we can't just destroy them and recreate them.
closed 2017-12-08 by canboat