opencpn-radar-pi/radar_pi #39: Use of on and off buttons is not consistent with other buttons

douwefokkema, 2017-11-24:
A button shows the current value of a parameter. When you want to change that current value press the button, you toggle it or get a sub-menu to change the value. On and Off are an illogical exception, these buttons show the value you will get after you press it. For consistency better make one button toggling on/off.

canboat, 2017-11-24:
You are right that not all buttons work the same way, see your confusion in #41 where you did not realise the way the AIS/ARPA button worked.

With On/Off the situation is even worse for Garmin as some controls have a tristate “off/manual value/auto value + computed value”.

We have some where the Off/On is set in the ‘control value’ resulting in a + - edit control to change it and some are toggles.

Altogether I agree there is room for improvement. What we could really use is that we draw the controls ourselves instead of using wxWidgets buttons. This would allow us to add colour (the Navico plotters use a triangular wedge in the top right of a button to show whether a setting is on or off instead of the text On / Off). it would also improve display on GTK (Linux) as there the button texts are left aligned instead of centered. The problem is that it is less beautiful on macOS as there is no good high resolution support in wxWidgets 3.1... 

I remember we had the same discussion a few years ago w.r.t. power (transmit / standby).

We tried the way the plotters do it (you get a Transmit button if the radar is in standby and a Standby button if it is transmitting) but I remember that most people did not like it (I did though…) I don’t mind the current “Power” menu as it gives a place to put the info button without getting into the way.

trudK45, 2017-11-24:
I agree. A bad example for the Garmin is Noise rejection "Button". It is not very nice operating On/Off in a sub menu with - and +. It should be a toggle button.

When it comes to what text is shown it becomes a little complicated. If it says On clicking the button naturally would set value to off and vice versa. But if the button can have more stats this is no longer feasible as you do not know what the current state is..

In that case you will need more text on the button. One showing the present state and one that shows you what will happen when you click the button.

trudK45, 2017-11-24:
After thinking a little more about this........   I would like to see for the buttons with not only On and Off is a divided button. The left part not being a button but a field telling you what state the function is in and the right part being the real button with text in it telling you what it will do. Problem might be size of text.

douwefokkema, 2017-11-24:
On / off  button could read: Power On / Off
                                                On
See also issue #41 on the function of our buttons.
For buttons with 3 states (true / relative / off for trails) we can still toggle, for more states we need a sub- menu.  We could also change the function of the buttons, but will be difficult to find a better solution.

canboat, 2017-11-24:
> On / off button could read: Power On / Off
> On
> 

I think the only solution that is consistent is that *all* buttons have two lines with the top line describing the function (but not possible states) and the 2nd line the current state (and not a verb!) We remove all verbs from the first line such as “Show AIS/ARPA"

So we get (some examples for both Navico and Garmin xHD):

Gain
Auto

Gain
Auto Low 33%

Power
Transmitting

Sea
Off

Noise Rejection
Off

AIS/ARPA
Shown    (and not Show, which is a verb!)

That way it works no matter how many different states there are for a particular control.

A complicated control will show the Edit menu (just like now). Maybe we can improve the edit menu, but I’m pretty happy with that part. 

What we could do is as Hans suggested is do ‘toggles’ for 2 or 3 states without the edit dialog (if there is only a limited set of changes, for instance Off/On or even Off/Low/Medium/High)

douwefokkema, 2017-11-24:
OK for me. Toggling works fine for at least 3 values, as is done with trails, toggling relative / true / off. I think up to 4 values can be toggled if it concerns increasing or decreasing a value.

Hakansv, 2017-11-24:
OK

canboat, 2017-11-24:
I don’t have time until Sunday to do anything, if somebody else wants to tackle some stuff go ahead.

Hakansv, 2017-11-24:
We're not in a hurry and I'm of for a visit to my daughter over the weekend. There are more in life then radars, obviously. :-)

canboat, 2017-11-27:
Fine. I will re-investigate whether it is possible to have multiple fonts in 1 button, probably not (at least not on all 3 platforms) which would make it possible to have a visual difference between purpose and state.

canboat, 2017-11-27:
What Hans suggested with left/right is definitely not possible with standard wxWidgets buttons.

douwefokkema, 2017-11-30:
Implemented in a44a53f3. The power menu now has a single power status button toggling to Standby / Transmit but still showing timed Tx info.

Hakansv, 2017-11-30:
Works fine. Nice feature: "On for Targets" 👍

douwefokkema, 2017-12-01:
I was not yet happy with the text "Power Status" on the power button. So I changed it to "Transmit / Standby" (as it was in earlier versions), this better indicates it is a button that can be pushed. As an extra I swap it to "Standby / Transmit" depending on the state.
Better idea's? Reopen this one or change it in the code.

canboat, 2017-12-02:
Douwe, first you agree to this statement (guideline, mantra, whatever you want to call it):

>  *all* buttons have two lines with the top line describing the function (but not possible states) and the 2nd line the current state (and not a verb!) 

and then your first change after that is to change the Power button to something that is not according to that guideline. Standby and Transmit are possible states for the radar, and do not describe the function itself. So sorry, but I do not agree.

douwefokkema, 2017-12-02:
I think the function is very clear this way. "Standby / Transmit"  describes a function, you can change from one to the other. Everyone will understand the function of the button. "Power" is wrong as we do not do anything with the power. And many users will not understand what "Radar state" will do. But be welcome to implement something better. But take my name off from this one.

canboat, 2017-12-02:
> "Standby / Transmit" describes a function, you can change from one to the other.

*ALL* other functional buttons (clear cursor, delete targets, etc.) do not have a current state in a 2nd line. This is what I am trying to make clear: a button with a state should have a description in the first line that does not describe the current state. 

Don’t you think it is confusing and repetitive that it says

Standby / Transmit
    Transmit

and

Transmit / Standby
    Standby

I currently have replaced it with ‘Radar’. I tried 'Antenna' but that looked wrong. I agree that this is a hard one and that 'Power' doesn't really fit either because we're not switching it off (Although we could, there is a real 'power off' command for the Navico radars that we do not implement (yet.)) Please take a look. 

I have also changed the state descriptions when timed standby is selected. The states are now:

Standby
Transmit
Standby for mi:ss
Transmit for mi:ss
Transmit for targets

so that the first word always describes the current state, not the action that will take place in minutes. I think that is much clearer.

canboat, 2017-12-02:
Also changed are:

Off / Relative / True trails -> Trails motion
Target trails -> Trails duration

which is also an improvement i.m.o.

Hakansv, 2017-12-02:
My opinions for info:
- "Power" is not correct, agreed. 
- Transmit/Standby is relevant. Once a new user has learned that these dialog rectangles are buttons and not only info texts everything would be clear. 
- Trail "motion" is better. My sense for English is though not good enough to fully assess that expression, but I'll translate to Swedish using a word equivalent to "behavior".

When now discussing UI. Again i.m.o. :
- The button "Cursor" would be shown only when a PPI is shown.
- The "AIS/ARPA on PPI" would also be shown only when a PPI is shown. (The text "on PPI" may describe enough but if also hidden while no PPI no doubts at all.

canboat, 2017-12-02:
>  Once a new user has learned that these dialog rectangles are buttons and not only info texts everything would be clear.

I think part of the issue here is that the Windows 10 GUI is so 'flat' and 'square'. In Windows 7 it's much more obvious these are buttons.
[image]
[image]

How about this: if you have a radar detected but in standby the menu is now very short, it only shows 3 buttons (as there is no point to show the adjust/advanced/view/cursor/guard zone buttons). Maybe it should have a text

Your radar is <status>
Press the 'Radar' button
to switch it on

(If the status is other than Off.)

>  The button "Cursor" would be shown only when a PPI is shown.

Agreed. 

>  The "AIS/ARPA on PPI" would also be shown only when a PPI is shown. (The text "on PPI" may describe enough but if also hidden while no PPI no doubts at all.

Agreed.

Hakansv, 2017-12-02:
The tiny menu is fine when in standby but i do agree to:
_Your radar is
Press the 'Radar' button
to switch it on_

trudK45, 2017-12-02:
Actually, I think this needs to be worked upon a lot more.
I think you, who have worked with the code might not be the best source of opinion.
You are somewhat biased by the fact that, for you everything is simple/natural.

To wish for would be some individual that has worked with UI for complex systems.
I can ask around. I do not know anyone strait away bet there are people on one of the
companies I have worked for that surely know how a good UI should look like.
Of course I do not know if he/she would be willing to participate in this project.
Maybe we could ask the forum ?

I also, have opinions on the menus but I have kept from revealing those at the moment as I
think it's more important to get the radars functioning.

I think we need a parallel discussion about the appearance. Otherwise this is greatly going to interfere
with the ongoing development.

To summarize, lets go on with development and bug-fixes and take a discussion how the looks of things should be separately.

Hakansv, 2017-12-02:
Tested current changes to Emulator only but looks good so far. I'll take ride to the boat tomorrow for BR24 tests.
@Hans, I do agree someone not deeply involved would give opinions about the UI but maybe not someone not used to sit alone i a little rocking boat, by a small screen, doing the best to gather all info on screen and around to make a safe decision. It's not an office or a trade ship. People like us would be the experts here.

closed 2017-12-08 by canboat