opencpn-radar-pi/radar_pi #40: Timed transmit should stay on when an ARPA target is detected

douwefokkema, 2017-11-24:
ARPA targets detected in a guard zone will get lost after the Tx period ends. Targets might pass without an alarm through the guard zone.

canboat, 2017-11-24:
Good idea.

Hakansv, 2017-11-24:
I may not understand but if no guard alarm is checked but only ARPA ON the timed transmit is not interrupted. It's only interrupted by a guard alarm - or? (I have no moving targets, can't check)
And if the guard alarm is checked the TimedTransmit would end and start transmission if there's a target in the zone. Irrespective of ARPA ON is checked or not.

douwefokkema, 2017-11-30:
Håkan: to avoid returns from birds or sea-state raising an alarm you might only use ARPA and not guard alarm. And targets may move outside the zone and still be dangerous.

I think this item should not be seen as an enhancement but as a necessary condition to use ARPA with timed transmit. Implemented in a44a53f3ac.

Hakansv, 2017-11-30:
OK, convinced.

douwefokkema, 2017-12-01:
Tested by Håkan I saw somewhere. To be reopened if someone disagrees.

closed 2017-12-01 by douwefokkema