opencpn-radar-pi/radar_pi #41: No targets shown in PPI window for 4G

douwefokkema, 2017-11-24:
Targets shown on overlay and in Garmin PPI windkow don't show in 4G window. Contours of these targets do show in the 4G window.
[image]

douwefokkema, 2017-11-24:
In case we only run the 4G radar, targets are shown as they should.

canboat, 2017-11-24:
The menu button shows that you have not enabled AIS/ARPA — the button says “Show AIS/ARPA” in your screen shot which means that it is not enabled now.

douwefokkema, 2017-11-24:
Then the text on the button is wrong or at least confusing. 
Buttons have two functions: indicate what it does and display the current value. This button should read:
"Show / hide AIS", and on the second line "Show"

douwefokkema, 2017-11-24:
Further discussed in #39

closed 2017-11-24 by douwefokkema