opencpn-radar-pi/radar_pi #42: "Hide radar" context menu not needed

douwefokkema, 2017-11-24:
The "Hide radar" context menu is not not needed and can be removed. If you don't want to see radar just switch off radar or turn overlay off and close the PPI window and menu (in this case both ARPA and trails continue to be recorded). The (irritating) warning info box "radar not seen" should only be shown when trying to switch on radar and should have a "close" button.

canboat, 2017-11-24:
Hm, that is a matter of opinion. I actually like the ability to hide all radar stuff with 1 click, just like AIS (+ (M)ARPA) targets.

I also use it to hide all radar windows before I go into preferences (but that is because the layering of the windows is not as good in macOS as on Windows, I think due to restrictions in wxWidgets 3.0.)

douwefokkema, 2017-12-01:
Doesn't matter that it is there, as long as it is used.

closed 2017-12-01 by douwefokkema