opencpn-radar-pi/radar_pi #48: Linux build error

Hakansv, 2017-12-01:
Hi. I've tried to build in Linux. Some errors occured:

[ 35%] Building CXX object CMakeFiles/radar_pi.dir/src/RadarFactory.cpp.o
In file included from /home/pi/Builds/radar_pi/src/RadarType.h:81:0,
                 from /home/pi/Builds/radar_pi/src/RadarFactory.cpp:33:
/home/pi/Builds/radar_pi/src/garminhd/garminhdtype.h:48:1: error: macro "DEFINE_RADAR" requires 8 arguments, but only 7 given
 )
 ^
In file included from /home/pi/Builds/radar_pi/src/RadarType.h:82:0,
                 from /home/pi/Builds/radar_pi/src/RadarFactory.cpp:33:
/home/pi/Builds/radar_pi/src/garminxhd/garminxhdtype.h:48:1: error: macro "DEFINE_RADAR" requires 8 arguments, but only 7 given
 )
 ^
In file included from /home/pi/Builds/radar_pi/src/RadarType.h:84:0,
                 from /home/pi/Builds/radar_pi/src/RadarFactory.cpp:33:
/home/pi/Builds/radar_pi/src/navico/br24type.h:36:1: error: macro "DEFINE_RADAR" requires 8 arguments, but only 7 given
 )
 ^
In file included from /home/pi/Builds/radar_pi/src/RadarType.h:86:0,
                 from /home/pi/Builds/radar_pi/src/RadarFactory.cpp:33:
/home/pi/Builds/radar_pi/src/navico/br4gatype.h:38:1: error: macro "DEFINE_RADAR" requires 8 arguments, but only 7 given
 )
 ^
In file included from /home/pi/Builds/radar_pi/src/RadarType.h:87:0,
                 from /home/pi/Builds/radar_pi/src/RadarFactory.cpp:33:
/home/pi/Builds/radar_pi/src/navico/br4gbtype.h:40:1: error: macro "DEFINE_RADAR" requires 8 arguments, but only 7 given
 )
 ^
In file included from /home/pi/Builds/radar_pi/src/RadarType.h:89:0,
                 from /home/pi/Builds/radar_pi/src/RadarFactory.cpp:33:
/home/pi/Builds/radar_pi/src/navico/haloatype.h:38:1: error: macro "DEFINE_RADAR" requires 8 arguments, but only 7 given
 )
 ^
In file included from /home/pi/Builds/radar_pi/src/RadarType.h:90:0,
                 from /home/pi/Builds/radar_pi/src/RadarFactory.cpp:33:
/home/pi/Builds/radar_pi/src/navico/halobtype.h:38:1: error: macro "DEFINE_RADAR" requires 8 arguments, but only 7 given
 )
 ^
In file included from /home/pi/Builds/radar_pi/src/RadarType.h:94:0,
                 from /home/pi/Builds/radar_pi/src/RadarFactory.cpp:33:
/home/pi/Builds/radar_pi/src/emulator/emulatortype.h:37:1: error: macro "DEFINE_RADAR" requires 8 arguments, but only 7 given
 )
 ^
In file included from /home/pi/Builds/radar_pi/src/RadarType.h:81:0,
                 from /home/pi/Builds/radar_pi/src/RadarFactory.cpp:33:
/home/pi/Builds/radar_pi/src/garminhd/garminhdtype.h:41:1: error: ‘DEFINE_RADAR’ does not name a type
 DEFINE_RADAR(RT_GARMIN_HD,                                /* Type */
 ^
In file included from /home/pi/Builds/radar_pi/src/RadarType.h:82:0,
                 from /home/pi/Builds/radar_pi/src/RadarFactory.cpp:33:
/home/pi/Builds/radar_pi/src/garminxhd/garminxhdtype.h:41:1: error: ‘DEFINE_RADAR’ does not name a type
 DEFINE_RADAR(RT_GARMIN_XHD,                                /* Type */
 ^
In file included from /home/pi/Builds/radar_pi/src/RadarType.h:84:0,
                 from /home/pi/Builds/radar_pi/src/RadarFactory.cpp:33:
/home/pi/Builds/radar_pi/src/navico/br24type.h:29:1: error: ‘DEFINE_RADAR’ does not name a type
 DEFINE_RADAR(RT_BR24,                                     /* Type */
 ^
In file included from /home/pi/Builds/radar_pi/src/RadarType.h:86:0,
                 from /home/pi/Builds/radar_pi/src/RadarFactory.cpp:33:
/home/pi/Builds/radar_pi/src/navico/br4gatype.h:31:1: error: ‘DEFINE_RADAR’ does not name a type
 DEFINE_RADAR(RT_4GA,                                      /* Type */
 ^
In file included from /home/pi/Builds/radar_pi/src/RadarType.h:87:0,
                 from /home/pi/Builds/radar_pi/src/RadarFactory.cpp:33:
/home/pi/Builds/radar_pi/src/navico/br4gbtype.h:33:1: error: ‘DEFINE_RADAR’ does not name a type
 DEFINE_RADAR(RT_4GB,                                      /* Type */
 ^
In file included from /home/pi/Builds/radar_pi/src/RadarType.h:89:0,
                 from /home/pi/Builds/radar_pi/src/RadarFactory.cpp:33:
/home/pi/Builds/radar_pi/src/navico/haloatype.h:31:1: error: ‘DEFINE_RADAR’ does not name a type
 DEFINE_RADAR(RT_HaloA,                                        /* Type */
 ^
In file included from /home/pi/Builds/radar_pi/src/RadarType.h:90:0,
                 from /home/pi/Builds/radar_pi/src/RadarFactory.cpp:33:
/home/pi/Builds/radar_pi/src/navico/halobtype.h:31:1: error: ‘DEFINE_RADAR’ does not name a type
 DEFINE_RADAR(RT_HaloB,                                        /* Type */
 ^
In file included from /home/pi/Builds/radar_pi/src/RadarType.h:94:0,
                 from /home/pi/Builds/radar_pi/src/RadarFactory.cpp:33:
/home/pi/Builds/radar_pi/src/emulator/emulatortype.h:30:1: error: ‘DEFINE_RADAR’ does not name a type
 DEFINE_RADAR(RT_EMULATOR,             /* Type */
 ^
CMakeFiles/radar_pi.dir/build.make:302: recipe for target 'CMakeFiles/radar_pi.dir/src/RadarFactory.cpp.o' failed
make[2]: *** [CMakeFiles/radar_pi.dir/src/RadarFactory.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/radar_pi.dir/all' failed
make[1]: *** [CMakeFiles/radar_pi.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2

trudK45, 2017-12-01:
I am at it. Son ok i hope. Did not show up as an error on WIN only as a warning.

trudK45, 2017-12-01:
Håkan, try now.

Hakansv, 2017-12-01:
Hans..
Yes! Builds fine both in Win and Linux(RPi-2B Rasbian). Well done.
I am curious to test the new radar_pi on my Rpi.
Thanks

closed 2017-12-01 by Hakansv