opencpn-radar-pi/radar_pi #54: Graurd zone is printed on screen even after radar Standby

Hakansv, 2017-12-04:
At least in emulator mode. Will check live soon.
Two pictures: Transmit/Standby
[image]

[image]

canboat, 2017-12-04:
Yes, this is an experiment, on purpose, so that you see where the guard zone will be if the radar is on (as the guard zones are now no longer ‘explained’ on the top menu.)

Click the radar icon in the toolbar to hide all radar stuff.

(We can always change this if we don’t like this change.)

Hakansv, 2017-12-04:
OK, understood. Maybe not bad, at least when TimedIdle is active, Hmm..
It's closed not only by the icon but also by the right click menu item "Hide Radar" - good!
BTW:  The texts "Hide Radar" and "Show Radar"  would read: "Hide Radar Control" / "Show Radar Control" ??

canboat, 2017-12-04:
> On 4 Dec 2017, at 10:39, Håkan Svensson <notifications@github.com> wrote:
> 
> OK, understood. Maybe not bad, at least when TimedIdle is active, Hmm..
> It's closed not only by the icon but also by the right click menu item "Hide Radar" - good!
> BTW: The texts "Hide Radar" and "Show Radar" would read: "Hide Radar Control" / "Show Radar Control" ??
> 
> 
No those are different, they control whether you see the menu. I’m at the office so I can’t check right now when we show those in the context menu. I think we put like 5 items in the OpenCPN context menu

Hakansv, 2017-12-04:
Again, the OCPN context menu item Show/Hide Radar does actually show/hide also guard zones ad the radar icon. That's good I think. But if I close the radar menu by the Windows [X], guards zone(s) are not hid. It would be consequent if they did, i.m.o.

canboat, 2017-12-04:
Nah, not in my opinion. I like running with the menu's hidden, although I usually do have the PPI window(s) open.

Hakansv, 2017-12-04:
I don't see the purpose to have guard zone(s) on screen when the radar is switched off. It's disturbing the chart. But I can use two buttons to turn the radar off. Not a big problem.

trudK45, 2017-12-04:
Håkan!

You might want to set up the guard zone before you transmit and it is 
easier if you see what you are doinf.

/Hans

Den 2017-12-04 kl. 22:15, skrev Håkan Svensson:

>
> I don't see the purpose to have guard zone(s) on screen when the radar 
> is switched off. It's disturbing the chart. But I can use two buttons 
> to turn the radar off. Not a big problem.

---
Detta e-postmeddelande har sökts igenom efter virus med antivirusprogram från Avast.
https://www.avast.com/antivirus

Hakansv, 2017-12-04:
You'll see what you're doing also before activating a guard alarm or ARPA.
While you edit the zone options inner/outer/start/end the corresponding zone is viewed a short while. 
Håkan

canboat, 2017-12-05:
I added this because we no longer show the state of the Guard zones on the main menu.

I am happy either way:

- remove them again in standby, go back to old way
- leave it as it is now
- make this user-settable in preferences…

Kees

Hakansv, 2017-12-05:
Kees, nice!
If others agree the user-settable would be fine for me.
Thanks

trudK45, 2017-12-05:
User settable is ok for me, On as default.
/Hans

Den 2017-12-05 kl. 11:07, skrev Håkan Svensson:

>
> Kees, nice!
> If others agree the user-settable would be fine for me.
> Thanks

---
Detta e-postmeddelande har sökts igenom efter virus med antivirusprogram från Avast.
https://www.avast.com/antivirus

trudK45, 2017-12-05:
When down in Gurard zone menu and pushing back you go up 2 levels. Should be only one i think.

Hakansv, 2017-12-05:
Two levels back confirmed here as well. I agree to the suggestion.

trudK45, 2017-12-05:
Would it be possible to adjust the guard zone with the mouse ?
Drag the radius and ankle ?
That would be pretty cool.

canboat, 2017-12-08:
Adjusting with the mouse is something we tried a while, but the problem is we don't really get a 'drag' event I think, so adjusting it becomes tiresome with lots of context menu clicks. In the end, we dropped this because I could not be bothered fixing it after it broke again.

commit d9c5cb29 (2017-12-08): Fix #54. Make display of stuff on overlay optional. Default is on.
closed 2017-12-08 by canboat