opencpn-radar-pi/radar_pi #57: Timed transmit can't be swiched on.

Hakansv, 2017-12-04:
The plus/minus buttons don't work

trudK45, 2017-12-04:
Confirmed on xHD as well.

canboat, 2017-12-08:
This should work now, but I haven't checked with a live radar yet so it is hard to know whether it really does.

Hakansv, 2017-12-08:
Yes the +- and Off buttons now works and when switched on the initial transmit period ends correct but after that a continuously shift between standby-transmit occurs.

trudK45, 2017-12-08:
Same here. 
The intended transmit period seem quite short (30 sek). The spin up takes 10-15 sec. There is then even less time for transmitting. We could measure the time from command to transmit and start decrementing the transmit timer when we reached transmitting. Then subtract the standby time with the same amount so that the spin up time is part of the standby time.

canboat, 2017-12-10:
> The intended transmit period seem quite short (30 sek). The spin up takes 10-15 sec. There is then even less time for transmitting. We could measure the time from command to transmit and start [...]

Only an issue with magnetron radars, obviously, as they require spin up. An advantage of the newer pulse compression and older FMCW radars that have fully electronic transmitters is that they don't spend 90 seconds warming up.

canboat, 2017-12-10:
Do we want to keep the short transmit times? The Garmin hardware feature cannot transmit for less than a minute, so if we want the UI to be the same across multiple radars then we need to make the 'timed run' values a value from 1 to 99 minutes as well as 'timed idle'.

trudK45, 2017-12-10:
I have been thinking about this and come to the conclusion that there is no good reason to utilize the hardware function of timed transmit in the radar. I can only find one reason and that is when you have a Garmin plotter on the side. If you activate timed transmit on the plotter OCPN will be out of sync.

douwefokkema, 2017-12-10:
I think one minute is a nice minimum transmit time. Just to lock a target and get a reliable target speed 10 radar revs is needed. Transmitting shorter than a minute is risky.

Hakansv, 2017-12-10:
Agree to minimum one minute. Make the options 1, 2, 3 minutes??

Hakansv, 2017-12-13:
Tested on Navico BR24 and emulator. Works fine except guard alarm is not deactivating. >> Own issue

closed 2017-12-13 by Hakansv