opencpn-radar-pi/radar_pi #59: Range control "Auto" button should stay gray when activated.

trudK45, 2017-12-05:
The heading says it all.

trudK45, 2017-12-08:
There is something fishy here. When pressing Auto, state goes to RCS_AUTO_1 but then goes to RCS_MANUAL for some reason and this results in m_auto_button->Enable() being executed in line 1327 of ControlsDialog.cpp turning the button on again. I am having difficulties determining where the state is set to RCS_MANUAL.

trudK45, 2017-12-08:
I found out where this happens.
It's when the range report from the radar is read in GarminxHDReceive.cpp line 520. There Update is called. Going into Update it always sets RCS_MANUAL.
This must be wrong but I do not know how to correct this yet.

canboat, 2017-12-08:
I know an elegant way, I will work on this.

canboat, 2017-12-08:
It works for me now. 

There is one thing -- the auto state is not persisted (it never was) so if you have overlay on and restart O, it reverts back to Auto Range.

trudK45, 2017-12-08:
Works fine. We can close this one. When I write Issue # some other issues come up but not #59.

closed 2017-12-10 by canboat