opencpn-radar-pi/radar_pi #60: Auto range does not cover the whole chart.

trudK45, 2017-12-05:
If you pan the chart so that the boat is at the far right on the screen you have no radar coverage on the on the left of the screen. The range on this plot is set to 4 Nm and the distance to the left of the chart is > 7 Nm.

[image]

trudK45, 2017-12-08:
Maybe this is how we want it ?

canboat, 2017-12-08:
I think this is what my plotter does, so I am not against leaving it like this.

trudK45, 2017-12-08:
Ok, this is how the Garmin plotter seem to work as well. The original gradar_pi and my gxradar_pi works differently. There the range is chosen so that the range covers the chart.
I think it's ok to have it as it is now. I will close the issue. Any other comments ?

canboat, 2017-12-08:
Works as designed, I would say.

closed 2017-12-08 by canboat