opencpn-radar-pi/radar_pi #61: No ring at extreme range.

trudK45, 2017-12-05:
Activating the option under preferences  "Show ring at extreme range" there is no ring.

Hakansv, 2017-12-05:
On BR24 and emulator I've a ring. You don't use OCPN distance rings do you? For me they often cover the radar ring.
The pointer was on the ring at shot event. 
[image]

trudK45, 2017-12-05:
On Garmin xHD there is nor ring on the chart. There is on o the PPI .

trudK45, 2017-12-08:
Could this have to do with the fact that the number of spokes vary on the Garmin ?
I would suspect so looking at the code.

canboat, 2017-12-08:
I do see a ring, on both PPI and overlay. Don't get confused with the image, it is a radar image from Honduras overlayed onto Holland :-)

[image]

trudK45, 2017-12-08:
What spoke length are you running with ?

canboat, 2017-12-08:
It's your file!

trudK45, 2017-12-08:
Well, I can't see any ring here. I have activated the Show ring at extreme range". I have Overlay transparency = 20 %. I am transmitting. Is it meeded ? Is there anything else needed ?

canboat, 2017-12-08:
Nope, that should be enough. It can be quite far away ...

trudK45, 2017-12-08:
I have radar range = 1 Nm and scale on chart so I can see 48 Nm. No ring.
No ring if I rescale so I can see 5 Nm. Ring should be a small distance outside "No transmit zone"

trudK45, 2017-12-08:
No, that is not enough. You have to have set "Show target trails on overlay" also.
This must be wrong.

commit b5dd0afa (2017-12-08): Fix #61.
closed 2017-12-08 by canboat