opencpn-radar-pi/radar_pi #62: Menu "Display Transparency" +  - changes value 10%.

trudK45, 2017-12-05:
Clicking plus or minus makes a change of 10%. Either change the step or the button to reflect the change.

Hakansv, 2017-12-05:
Observation confirmed but I think it's relevant. "+" for increase and "-" for decrease.  10% step is OK for this function. What's the problem?

trudK45, 2017-12-05:
We have a + - 10 button. Why not use that instead of + - 1 ? Then it would be clear that you only can move it at increments of 10. To have the same button here that in all other menus moves the value by 1 and have this as an exception feels wrong.

Hakansv, 2017-12-05:
Ahaa! Sounds fair, of possible to view the +10 sole.

trudK45, 2017-12-06:
Everybody agree ?

canboat, 2017-12-06:
Yes.

If you want to try changing this — it’s pretty simple. EnterEditMode() or whatever can look at the m_ci object and look at the step (and unit!) so if the m_stepValue is > 1 then add that to the button text. And maybe unit as well. So the idea is you keep showing the “+1” and “-1” buttons just change the labels on the buttons.

canboat, 2017-12-06:
> On 6 Dec 2017, at 14:54, Kees Verruijt <kees@verruijt.net> wrote:
> 
> Yes.
> 
> If you want to try changing this — it’s pretty simple. EnterEditMode() or whatever can look at the m_ci object and look at the step (and unit!) so if the m_stepValue is > 1 then add that to the button text. And maybe unit as well. So the idea is you keep showing the “+1” and “-1” buttons just change the labels on the buttons.

that way it will also work correctly if we ever have a control that changes in steps of 5 or whatever not equal to 1 or 10.

Hakansv, 2017-12-06:
Yes, agree!

trudK45, 2017-12-07:
I know where to find the stepValue but having a hard time figuring out where to change the comment.
Is it button->m_comment or somewhere else ?
Should this be for CT_TRANSPARENCY only or for all ?
The transparency should have a %-sign added.

trudK45, 2017-12-07:
Ok, I now have this working. If the %-sign is added here it should be added elsewhere also.
To be consistent it should be on all + - except for Range.
What do you say ? Should we have the %- sign.
It think it look rather nice.

[image]

Hakansv, 2017-12-07:
Very nice and informative. Of course we can have it like that. Once you've started I think you would like to finalize with the others? Simple signs like % would fit on all OPs I think.

trudK45, 2017-12-07:
Added %-sign to appropriate buttons. Issue #62

canboat, 2017-12-08:
You could have made better use of `m_ci.unit` and thus avoid having to hardcode control types here.

commit 3bb0c45d (2017-12-08): Fix #62. Use m_ci.unit, this allows removal of hardcoded control types.
closed 2017-12-08 by canboat