opencpn-radar-pi/radar_pi #65: Distance rings in distance in PPI is wrong.

trudK45, 2017-12-09:
When you specify inner and outer distance in the the PPI they do not align with distance rings.
The Guard zone seems correct. The Guard zone inner border is set to 3 and the outer to the distance to a ship. The ships distance can be seen in the AIS Target Query and corresponds with outer the guard border.
I.e. AIS and Guard zone distance correlate but not the radar distance rings. This seems to be the case on all ranges.

[image]

canboat, 2017-12-10:
Does this work for you now?

trudK45, 2017-12-10:
Yes this is now good. One thing though. If you shut down OPCN with guard zone active, then when you start OPCN the guard zone does not show. Not a big deal but it does not get back to the same state. You have to start transmitting before the guard zone appears.

canboat, 2017-12-10:
> On 10 Dec 2017, at 15:00, trudK45 <notifications@github.com> wrote:
> 
> Yes this is now good. One thing though. If you shut down OPCN with guard zone active, then when you start OPCN the guard zone does not show. Not a big deal but it does not get back to the same state. You have to start transmitting before the guard zone appears.
> 
> 

When the plugin has no idea of the radar range (and is never told it) it refuses to draw anything, right now. I can fix this.

trudK45, 2017-12-10:
Is it not told by the received packets from the radar ?

canboat, 2017-12-10:
It was looking at the 'real' radar range received, e.g. the spoke data. Fixed that now.

commit fae3bff4 (2017-12-10): Fix #65. Show guard zone even when not transmitting.
closed 2017-12-10 by canboat