opencpn-radar-pi/radar_pi #68: ARPA detection is not working since some of the last commits

Hakansv, 2017-12-10:
Tested with Emulator and 4GA radar. Neither works. Guard alarm is still working.

canboat, 2017-12-10:
It works fine for me with origin/master, i.e. without your pull request.

A common pitfall is to disable AIS/ARPA on the PPI display, which means you don't see the detected targets on the PPI.

Hakansv, 2017-12-10:
1. Yes I'm on origin/master wo changes. Latest pull 5 minutes ago. 
2. It's not about PPI. (I use overlay). There's no ARPA contour at all so the detection process is not started. 
I'm not sure how to trace this. Any advice?

canboat, 2017-12-10:
The first thing to do is to see if you don't see any ARPA contours in the PPI either. (I see both...)

Second is to try to reproduce it with a 'well known pcap', e.g. one in the examples directory.

Hakansv, 2017-12-10:
As said, no contours. Nether in PPI nor overlay.
And, Yes I use the same pcap(s) as for the last two years. Radar type is set to 4G A as usual.
The ARPA detection was there two-three days ago. Also in the emulator. Neither works now.
I'm on Win10 also as usual.

canboat, 2017-12-10:
Maybe Douwe can help, I don’t have this issue and I am busy working on timed transmit.

douwefokkema, 2017-12-10:
Just tried on origin/master, fae3bff4. All fine.

Hakansv, 2017-12-10:
OK thanks for testing.  Something on my (build?) system then.  Hmmm

douwefokkema, 2017-12-10:
I also tested your PR, just perfect, but not tested with AIS targets (don't have them here).
[image]

trudK45, 2017-12-10:
Here we got an island moving to the NW at 2.35 knots.

[image]

trudK45, 2017-12-10:
I am new to ARPA and so need to ask. Why are only certain targets registered ?
Do you discard targets that have AIS transmitters ?

[image]

trudK45, 2017-12-10:
By the way. The Radar symbol in the menu is now a zig-zag piece.

canboat, 2017-12-10:
> Do you discard targets that have AIS transmitters ?

Yes, nice no?

> The Radar symbol in the menu is now a zig-zag piece.

This is common if you run under a debugger without the correct directory structure. I just tested a real installer and then I did not see any issues with the icon.

canboat, 2017-12-10:
> Here we got an island moving to the NW at 2.35 knots.

Douwe?

trudK45, 2017-12-10:
I am running a real installer and still get the zig-zag.

canboat, 2017-12-10:
Can you try finding this yourself, I'm busy on timed transmit.

trudK45, 2017-12-10:
This was surprising. Reinstalled the plugin and now it's ok.

douwefokkema, 2017-12-10:
"Do you discard targets that have AIS transmitters ?"
Yes, when an ARPA target is closer than a certain limit to an AIS target the pi does not send the target to O, the contours are shown.

"Here we got an island moving to the NW at 2.35 knots."
On the island there are multiple unstable blobs. The ARPA system can't make much sense out of these blobs. If the radar would show the island as one stable blob there would be no problem. ARPA targets do need some kind of human interpretation.

Hakansv, 2017-12-12:
Found it!!!
This error was on one of my PCs only. So checked the config file for anomalous setting and found:
Radar0MinContourLength=33620738 where default setting should be 6. 
How and when the abnormal value was set I don't know. When corrected ARPA is working nice also on that PC. Sorry for any trouble for you but it may be a lesson learned?

closed 2017-12-12 by Hakansv