opencpn-radar-pi/radar_pi #74: Navico: Guard zone alarm does not deactivate timed transmit

Hakansv, 2017-12-13:
On BR24 and Emulator
Guard zone Alarm is not deactivating Timed transmit. Would go to "Off" when any zone alarms.
Guard zone ARPA is though working fine, “Transmit for Targets”

trudK45, 2017-12-15:
This is the way the code is written and maybe how we want it, or ?
It's not that difficult to change.

trudK45, 2017-12-15:
You could try inserting this in RadarInfo.cpp at about line 1483.
Just before declaration of :

  time_t now = time(0);
  int    time_to_go;

```
  int bogeys;
  
  for (int z = 0; z < GUARD_ZONES; z++) {
    bogeys = m_guard_zone[z]->GetBogeyCount();
    if (bogeys > 0 || (m_pi->m_guard_bogey_confirmed && bogeys == 0)) {
      return;
	}
  }
```

Hakansv, 2017-12-16:
"_This is the way the code is written and maybe how we want it, or ?_"
If so it's a change. It has been the desired function to switch off since the function was introduced. The logic is if there's an alarm I first check where the target is on screen and if close check outside if there's a need for immediately actions. Then I don't want the radar to shut down while I'm busy with other things and may need a quick look for the radar target. So, yes, I do think the timed transmit function would be turned off if there's an alarm. Alternatively use the same logic as for a detected ARPA target.

trudK45, 2017-12-16:
And when all targets disappear do we want the timed transmit to be reinvoked automatically or is it ok if that needs manual assistance ?

douwefokkema, 2017-12-16:
"And when all targets disappear do we want the timed transmit to be re-invoked":
Yes, I would say. So if there is an ARPA target detected, the radar stays on until it has disappeared again, then again timed transmit. No user interference when no (AIS) alarm was raised. You don't want to get out of your bed to switch on timed Tx again. And you would not even know you have to get out.

trudK45, 2017-12-16:
At the moment I have code that changes to continues transmit when a target is found in either ARPA or Guard zone. It does not return to timed transmit when targets disappear.
I could rewrite the code to fix this but I know Kees has ideas how to implement this so I am reluctant to go on with this.

Hakansv, 2017-12-17:
As long the timed transmit interrupts while an alarm is active I'm happy. To return to TT when gone would be OK but I think the pause period timer would be set to Time(0) when the alarm is gone, just to sure.

canboat, 2017-12-18:
> On 17 Dec 2017, at 18:43, Håkan Svensson <notifications@github.com> wrote:
> 
> As long the timed transmit interrupts while an alarm is active I'm happy.

OK

> To return to TT when gone would be OK but I think the pause period timer would be set to Time(0) when the alarm is gone, just to sure.

Not 100% sure I understand what you are saying here, can you rephrase? Is this a wish or what you think the code does right now?

canboat, 2017-12-18:
I’m not able to do any work because of work + social events this week. Christmas this weekend, so probably the same, so I’m more than happy for Hans to fix this.

trudK45, 2017-12-18:
We are launching today in a few hours and will go north to Grenada so I wont be able to do anything for a couple of days. We will see when I get internet again.

canboat, 2017-12-18:
Bon voyage!

trudK45, 2017-12-18:
Merci.
One question Kees. I am thinking of introducing another Radar state. Something like RADAR_TT_PENDING when you have detected targets. Or replace the IDLE which is only used for hardware. Agree ?

canboat, 2017-12-18:
Yes, that seems OK.

Hakansv, 2017-12-18:
Kess..
It's a discussion about the new suggestion what's should happen to TimedTransmit, TT, after a zone alarm. 
- Before we just turned TT off when an alarm occurred.
- Now there's a discussion to return to TT when the alarm is no longer active, as for the ARPA. In that case I think it's important to wait to start Timed Idle a while after the alarm disappeared. Just to be sure the target was not hid by a wave or something similar. The waiting time can for example be a full transmit period. 

Quote:
Me: _To return to TT when gone would be OK but I think the pause period timer would be set to Time(0) when the alarm is gone, just to sure._
Kees:_Not 100% sure I understand what you are saying here, can you rephrase? Is this a wish or what you think the code does right now?_
/Quote

canboat, 2017-12-18:
Top posting should be forbidden :-) but I will continue the reading up from the bottom.

I agree, when TT is resumed because the ‘reason(s)’ (ARPA/targets, other discussion) resumes it should start with a transmit period (of whatever the transmit period was) not a standby (idle) period.

So the following could be a time-line with a 3 minute standby and 1 minute transmit:

0:00 radar goes standby (status “Standby for 180s”, counts down to 0)
3:00 radar starts transmitting (status “Transmit for 60s”, counts down to 0)
3:45 a target / ARPA is detected (=> TTX on hold, status “Transmit for ARPA” or “Transmit for targets”)
4:30 target disappears (=> status “Transmit for 60s”, counts down to 0)

> On 18 Dec 2017, at 11:49, Håkan Svensson <notifications@github.com> wrote:
> 
> Kess..
> It's a discussion about the new suggestion what's should happen to TimedTransmit, TT, after a zone alarm.
> 
> Before we just turned TT off when an alarm occurred.
> Now there's a discussion to return to TT when the alarm is no longer active, as for the ARPA. In that case I think it's important to wait to start Timed Idle a while after the alarm disappeared. Just to be sure the target was not hid by a wave or something similar. The waiting time can for example be a full transmit period.
> Quote:
> Me: To return to TT when gone would be OK but I think the pause period timer would be set to Time(0) when the alarm is gone, just to sure.
> Kees:Not 100% sure I understand what you are saying here, can you rephrase? Is this a wish or what you think the code does right now?
> /Quote
>

Hakansv, 2017-12-18:
Very good! Exactly according to my view at least.

Top posting is a "new" habit since mail was introduced, many years ago. We may be stuck to "old habits" but sometimes it's efficient. :-)

closed 2018-05-27 by Hakansv