opencpn-radar-pi/radar_pi #77: Raymarine support

BrannJoly, 2018-01-09:
Hi,
I was wondering if you had considered supporting Raymarine radars ? I have one, and I'm thinking about looking into it, but maybe there's a good reason why it hasn't been done before ? 

Cheers,
Brann

canboat, 2018-01-09:
None of the current developers have a Raymarine radar, so we can’t work on this.

Feel free to implement it yourself, it is now much easier than before. The most difficult part is analyzing the traffic between the radar and the MFD using an IP packet analyzer such as tcpdump. Once you have this, we’re willing to help you set up the code and fix bugs, but testing needs to be done by someone with access to the physical radar.

An alternative, if you are not a C++ developer, is to donate a Raymarine MFD and radar to one of the existing developers :-)

canboat, 2018-01-15:
See #79 .

closed 2018-01-15 by canboat