opencpn-radar-pi/radar_pi #78: Navico/Simrad Halo support

Hakansv, 2018-01-11:
Kees. Pls read this and following posts: http://www.cruisersforum.com/forums/f134/opencpn-radar-overlay-plugin-79081-53.html#post2552375

I'll invite Kevin to continue the discussion here instead. 
The Wireshark file he captured would be of interest to understand the Halo I suppose.

Kevin wrote:
 Hi Haken

I've attempted to capture a Wireshark file of both my wireless internet with NMEA 183 and my Ethernet Port with the radar data.

I slowly (about once every seconds ) went up the ranges from 91metres to 6 miles before the program crashed.
The problem at 3 Nm seems to coincide with a change of speed of the antenna.
the Halo Radar adjusts its speed when on auto according to the range. Even if I set the speed to slow on both displays the speed still changes to an extent and the problem is just the same.
I have also tried closing Halo B and just using Halo A = same problem.

I'm unable to upload the Wireshark file as an attachment due to size. Have you any suggestions?
__________________
Kevin Walters

KevinWalters1024, 2018-01-11:
Thanks Douwe and Hakan

I have joined 😀

Douwe, I assume you are referring to the ranges in the file haloatype.h and halobtype.h in the latest build. 

I am only in Greece with my radar until tomorrow. Then back out here in the middle of February when the boat will be out of the water. I should be able to do some further testing then 

Late March we are sailing it from Greece to England.

I will try to upload my  Wireshark capture later when I’m back on my main computer. 

Kevin

canboat, 2018-01-11:
Hi Kevin,

Welcome. As you can see github doesn’t pester you with advertisements like CF, and allows you to upload decently sized files — although it is probably smart to zip your pcap file to minimise the time you spend uploading.

Kees

KevinWalters1024, 2018-01-11:
Hi 

Attached is the Wireshark file. Please let ne know if its not in the correct format.

[Halo KevinWalters1024.zip](https://github.com/opencpn-radar-pi/radar_pi/files/1623860/Halo.KevinWalters1024.zip)

KevinWalters1024, 2018-01-11:
Hi Kees
Sorry my first Wireshark capture is useless to you as its only got the NMEA data on it from the wireless port.

This one should help as its got the spoke data from the Ethernet coming out of the RI-12 interface.

Kevin

[](
[Halo KevinWalters1024 110118.zip](https://github.com/opencpn-radar-pi/radar_pi/files/1623997/Halo.KevinWalters1024.110118.zip)
url)

canboat, 2018-01-12:
Walter, your latest PCAP is great. 

On my Mac it doesn’t crash above 3 NM but it does show a completely strange picture. 

My guess is that Navico fiddle with the header bits again, and we need to look at what they changed in detail.

Amazing picture by the way, looks great!

Kees

canboat, 2018-01-15:
With your PCAP it now works fine for me, @KevinWalters1024. 

Once you have a chance can you verify that all ranges work correctly and that the on-screen range in the PPI window is the same as in your plotter? And can you do that again but with the plotter set to metric ranges? 

If you see any values sandwiched between slashes, for instance, /11000/, you have found a range that the plugin doesn't handle well. Please supply a .pcap file separately for each such range that is broken (this makes it much easier for us to see where it goes wrong.)

(You can also try fiddling with ```src/navico/haloatype.h``` and ```src/navico/halobtype.h``` yourself, the RANGE_XXX defines would need adjusting.)

keesverruijt, 2018-06-10:
No comments received. Assuming closed by this fix.

closed 2018-06-10 by keesverruijt