opencpn-radar-pi/radar_pi #79: Raymarine radar_pi support

sv-pizzazz, 2018-01-12:
@ab2nkovs created a Raymarine compatible fork over a year ago, source code available at https://github.com/ab2nkovs/RMRadar_pi.

I took his code and made some modifications to it so that it can compile under Windows. The plugin works! Many thanks to ab2nkovs!

I would like to work on improving his code and publishing a solid Raymarine radar plugin. Basically, in his code he has replaced br24Receive.cpp and br24Transmit.cpp with his own version called RMControl.cpp. The plugin works with digitar radars or analog radars through the E80 chartplotter. Display of the radar data works very well. Most of the control functions work as well but sometimes are slow or buggy. The best use of this plugin is that users of Raymarine E80, who up to know could only use RayTECH software (slow, buggy software) to display radar on a PC app can migrate to OpenCPN and use its superior charting, weather routing, etc. capabilities. I have a couple of questions on improving the software and I would appreciate your guidance on how to proceed:

- The radar sends heading information but the plugin does not recognize it. What do I do?

- When connected to an E80, in addition to the heading data, the E80 sends a lot of additional information such as position, AIS, DSC, weather, etc. Is it possible to use/modify existing plugins to obtain this information via Ethernet (vs a serial NMEA connection) or is this the wrong way to do it?

- Where can I get guidance on using Wireshark to better understand the commands that are passed between the radar and the plugin? It sounds very complicated.

- Can we open another development effort specifically for the Raymarine plugin? Should I do it? How does this become an official plugin?

Thank you for the help, I am just learning how to do these things.

Regards,
SV Pizzazz

canboat, 2018-01-12:
It is a pity @ab2nkovs did not announce this to a more general audience, he probably did a lot of work and it would be super to integrate this into opencpn-radar-pi… Maybe you/we can contact him and see if he is willing to work on the generic opencpn-radar-pi.

> On 12 Jan 2018, at 21:41, sv-pizzazz <notifications@github.com> wrote:
> 
> @ab2nkovs <https://github.com/ab2nkovs> created a Raymarine compatible fork over a year ago, source code available at https://github.com/ab2nkovs/RMRadar_pi <https://github.com/ab2nkovs/RMRadar_pi>.
> 
> 

> I took his code and made some modifications to it so that it can compile under Windows. The plugin works! Many thanks to ab2nkovs!
>
> I would like to work on improving his code and publishing a solid [...]
You need to analyse where in the radar packets that data is hidden, and then use the same code as the 4G radar, which updates the plugin idea of heading as follows:

    bool radar_heading_valid = HEADING_VALID(heading_raw);
    bool radar_heading_true = (heading_raw & HEADING_TRUE_FLAG) != 0;
    double heading;
    int bearing_raw;

    if (radar_heading_valid && !m_pi->m_settings.ignore_radar_heading) {
      heading = MOD_DEGREES_FLOAT(SCALE_RAW_TO_DEGREES(heading_raw));
      m_pi->SetRadarHeading(heading, radar_heading_true);
    } else {
      m_pi->SetRadarHeading();
    }

> When connected to an E80, in addition to the heading data, the E80 sends a lot of additional information such as position, AIS, DSC, weather, etc. Is it possible to use/modify existing plugins to [...]
Yes, the E series sent a lot of data over TCP/IP I gather. You *could* invest a lot of time in that and convert those into NMEA injections. opencpn-radar-pi already does that for heading data (search for PushNMEABuffer…). 

The question is, how much effort do you want to invest in this and what will it bring? I take it your OpenCPN installation already gets most of this data? The reason we do heading is more that we want to rotate the chart the same way as the radar heading than producing a full NMEA0183 source for OpenCPN. 

> 
> Where can I get guidance on using Wireshark to better understand the commands that are passed between the radar and the plugin? It sounds very complicated.
> 
> 
It’s mainly a question of starting to understand what happens and isolating individual commands. At first you will be overwhelmed, but if you get the hang of it it becomes easier. My main trick is to create lots of individual pcap files that contain one single button press on the plotter, with 2-3 seconds after the button press so I can then analyse how the radar responds.
> 
> Can we open another development effort specifically for the Raymarine plugin? Should I do it? How does this become an official plugin?
> 
> 
The idea of github is that it is easy to fork and easy to integrate back in. You fork opencpn-radar-pi/radar_pi, add the Raymarine radar, and then send us a pull request. We then take that pull request and commit it to the main code. Our idea is that we have only a single plugin this way that works for all ‘end users’ without needing to download various plugins. And the individual radar developers don’t need to bother with fixes to the non-radar-specific code, such as the ARPA, AIS display, separate PPI displays, etc.

canboat, 2018-01-12:
I just checked @ab2nkovs source code and he did a LOT of work!

sv-pizzazz, 2018-01-13:
Thank you for the quick replies. I am struggling what is the best way to proceed given my inexperience with VC++. The learning curve is steep in the beginning. It took me a day to set up my IDE and modify the code to compile and now when I look back, the actual work in porting the code to Windows was less than 10 min if you know what you are doing -- I definitely do not.

Given that the plugin works well as it is, I feel that my time is better spent on Wireshark to come up with the correct commands. May be someone else who is more experienced with VC++ can own the code. I see three main tasks given your reply:

1. Find the heading packet and modify the code to read the heading. Forget about the rest from the E80.
2. Add ARPA functionality
3. Work on the overall stability and commands. Interestingly, RayTech has the same stability issues (e.g. sometimes it will change the range immediately, sometimes it will hang, so it may have to do more with the E-series APIs being buggy rather than issues with the plugin.

Would you be willing to take over the forking and merging the code? Shall I just look for a volunteer?

Another, unrelated question is SiriusXM support. My main interest is in building a solution that enables sending internet derived weather information to the E-series by pretending to be a SiriusXM weather processor. I just like the way the E-series displays SiriusXM weather but do not want to pay for it when I am inshore. The reverse could be helpful to OpenCPN users. How about a plugin that interfaces OpenCPN to a SiriusXM receiver and displays the information the same way grib files are displayed? That could be very useful. I could use Wireshark to reverse engineer the protocol, should not be that difficult.

canboat, 2018-01-13:
> On 13 Jan 2018, at 07:19, sv-pizzazz <notifications@github.com> wrote:
> 
> Thank you for the quick replies. I am struggling what is the best way to proceed given my inexperience with VC++. The learning curve is steep in the beginning. It took me a day to set up my IDE and modify the code to compile and now when I look back, the actual work in porting the code to Windows was less than 10 min if you know what you are doing -- I definitely do not.
> 
> Given that the plugin works well as it is, I feel that my time is better spent on Wireshark to come up with the correct commands. May be someone else who is more experienced with VC++ can own the code. I see three main tasks given your reply:
> 
> Find the heading packet and modify the code to read the heading. Forget about the rest from the E80.
> Add ARPA functionality
> Work on the overall stability and commands. Interestingly, RayTech has the same stability issues (e.g. sometimes it will change the range immediately, sometimes it will hang, so it may have to do more with the E-series APIs being buggy rather than issues with the plugin.
> Would you be willing to take over the forking and merging the code? Shall I just look for a volunteer?
> 
I don’t think we’re really in the mood. We just did Garmin and Garmin xHD together with Hans, who is also not a programmer, and it was not effortless to do this. We’d really prefer someone with more skills to help us.
> Another, unrelated question is SiriusXM support. My main interest is in building a solution that enables sending internet derived weather information to the E-series by pretending to be a SiriusXM [...]
You’re barking up the wrong tree here — the radar plugin group, by accident, hails from Sweden and the Netherlands, and since that is outside the USA we have no SiriusXM. You could try CruisersForum.

sv-pizzazz, 2018-01-15:
Kees & team,

I just wanted to thank you all for the great work on the radar plugin. I have been playing around with the @ab2nkovs Raymarine plugin on Windows and it works great. I learned that the issues with slow controls were due to my router configuration (it is now fixed), so it works very well over both wired and wireless connections. The key is to configure to router for fast multicast. 

Basically, I think the Raymarine plugin is stable and production ready as it is, with the minor modifications necessary for compilation under Windows vs. Linux. I am learning how to use GitHub and a little bit of programming, so when I am ready I will do a fork as request integration into your general plugin.

Meanwhile, I have a couple of suggestions for the user interface which I will put in a separate thread.

canboat, 2018-01-15:
Maybe @BrannJoly is willing to help as well? @ab2nkovs did all the hard work of reverse engineering, so what you would need to do is take the knowledge he made, fork the new plugin, copy one of the existing radars (the Garmin is the 'simplest', and that does conversion of 1 bit to 8 bits so it is probably the most suited) to a new 'raymarine' directory and then merge the work of @ab2nkovs. Maybe @ab2nkovs also wants to join?

canboat, 2018-01-15:
I saw that @ab2nkovs has just forked this repo...

bigalnz, 2019-04-01:
I too would like to see this Raymarine support added.

keesverruijt, 2019-04-04:
In order to help you a little bit I added a new "raymarineDigital" folder, and basically bashed to GarminxHD code in there. I made a few initial changes so that the classes are initialized the way that a RayDig radome has -- you only have a "report address" to start out with and need to learn the radar's IP address from what it sends you in a report.

What remains to be done now is most of the work: take @ab2nkovs work, and adapt it to the new code base.

Please fork this repo, and checkout the raymarine branch, improve it, and send a pull request. I think it is best if you do this with small changes, so that everything you improve it a little bit & we can comment on it.

If you send a pull request with a Raymarine .pcap file in the examples directory I could take a look when I have time so that we can at least show the radar image.

sv-pizzazz, 2019-04-04:
OK, thank you. Now, I will take the raymarineDigital branch and populate it with the data from @ab2nkovs work then attempt to compile the whole plugin. First step, would be to get a working receive as you suggest. Then I can experiment with the control. As an added benefit, I have the program called RayTech that connects to digital radar and controls it. So, I think I can figure out the control sequences quite well, once I get the setup. Are you guys using the latest VS or VS2013? thank you.

Hakansv, 2019-04-04:
O v5.0 uses wxWidgets 3.1.2 and VS2017 for Windows.
Håkan

bigalnz, 2019-04-05:
Good luck to you guys. This will rock. Do I understand it connectly that the plug in as it stands now will display the Raymarine radar in OpenCPN but no controls? I have Quantum II.

keesverruijt, 2019-04-05:
It is highly unlikely that this will work with Quantum II. It may it may not, we don’t know. Since there is an intermediate generation that had “full color” and 8 bits per radar pixel my guess would be “no”. The older ones that this will address had 4 bits per pixel.

If you’ve got one, and are willing to program ….

closed 2022-04-08 by keesverruijt