opencpn-radar-pi/radar_pi #80: Radar plugin user interface enhancements

sv-pizzazz, 2018-01-15:
I would like to suggest a few improvements to the command interface, especially with regards to using on touch devices. My issues with the current interface are mainly around the menu structure which is quite deeply nested, especially when compared to dedicated chartplotters. My suggestions are:

1. Radar operation requires adjustment of many parameters to get the best picture. The current approach of -10, -1, +1, +10 works well but it takes time to go through the menus. Will it be possible to have a horizontal bar with five fixed positions, -10, -1, center button, +1, +10, where the center button can cycle among the most frequently used adjustments, namely gain, rain, FTC, Clutter. Auto could be a separate button on the right.

2. When guard zones are selected, the center button can cycle between the parameters of the guard zones, then use the four buttons to adjust. It will be much quicker to adjust this way instead of typing the numbers.

3. When the cursor is placed on the PPI screen, the menu for EBL/VRM/ARPA should show automatically. This is a minor point though, it works very well now.

4. It would be great to extend this 5-button horizontal bar to the rest of Open CPN, i.e. for stepping through the grib files/climatology, where it could be -24h, -3h, +3h, +24h. The current interface is optimized for a mouse and even then it is quite slow to move the mouse around on a moving boat.

I hope this is helpful.
Pizzazz

canboat, 2018-01-15:
We already fought a lot over the command interface and the current setup is a compromise amongst the current development crew. This is a reflection of the different attitudes and what we deem important, but also of the different hardware that we use. The UI needs to work on the smallest (I guess 1024x600 or so, widescreen netbooks) vertical format, HiDPi and huge screens, tablets, mouse- oriented and keyboard- oriented folks.

It also needs to work equally well for folks using only the radar overlay and folks (mainly) using the PPI windows.

So, it turns out UI design is much much harder than just a few suggestions. Also, don’t underestimate the amount of work that goes into the coding of the UI as it is not easy to code it such that (a) everything works, and works in a predictable and expected manner, and (b) adapts to all different radars with different controls.

Here are my initial thoughts of your suggestions.

> On 15 Jan 2018, at 08:06, sv-pizzazz <notifications@github.com> wrote:
> 
> I would like to suggest a few improvements to the command interface, especially with regards to using on touch devices. My issues with the current interface are mainly around the menu structure which is quite deeply nested, especially when compared to dedicated chartplotters. My suggestions are:
> 
> Radar operation requires adjustment of many parameters to get the best picture. The current approach of -10, -1, +1, +10 works well but it takes time to go through the menus. Will it be possible to have a horizontal bar with five fixed positions, -10, -1, center button, +1, +10, where the center button can cycle among the most frequently used adjustments, namely gain, rain, FTC, Clutter. Auto could be a separate button on the right.
> 
> 
This would be a five to six button wide control. Where do you then cycle through or select all the other controls? How do you suggest placing those in a rectangular window?

I really would like a horizontal, vertical or circular slider that has the correct size for use with tablets. Unfortunately that doesn’t exist in wxWidgets, so we would have to write our own version. There is a horizontal slider, but it is too small and fiddly with a tablet. In fact the entire windows we use now are really crap on some OS at night, as they don’t dim properly. So the real challenge would be to do a full OpenGL design of the entire control window.

If you want to challenge the basics of the current design then AT LEAST you should work out an entire proposal, with mock-ups that show what the effect would be.
 
Also note that if you are trying the current plugin you should probably use a Navico PCAP file so you start appreciating how many controls we have to place. It has settings for antenna height, interference rejection, local interference rejection, scan speed, side lobe suppression, target boost, target expansion, target separation and noise rejection that need placement in addition to the regular gain/rain/sea settings as well as all the ‘software’ controls such as antenna offset, guard zones, arpa, marpa, trails, etc.
> When guard zones are selected, the center button can cycle between the parameters of the guard zones, then use the four buttonsto adjust. It will be much quicker to adjust this way instead of typing the numbers.
I agree that the number editing for the guard zones is not ideal. I would suggest something where you select it by clicking on the PPI window directly. We used to have something like that but this was in the time when there were no PPI windows, and it relied on using the right mouse button which is useless with a table and it was removed. Another idea is the sort of circular control you get on Android when entering a date and time. 

I must say that on my Windows 10 Surface Pro 4 that the OS is pretty good in popping up the on screen keyboard at the right time, and this works pretty well for me.
> When the cursor is placed on the PPI screen, the menu for EBL/VRM/ARPA should show automatically. This is a minor point though, it works very well now.
> 
> 
We used to do this, but it turns out that you click on the PPI window much more often than you actually want to work with the cursor, so this was removed. I had this especially on my tablet. 
> It would be great to extend this 5-button horizontal bar to the rest of Open CPN, i.e. for stepping through the grib files/climatology, where it could be -24h, -3h, +3h, +24h. The current interface is optimized for a mouse and even then it is quite slow to move the mouse around on a moving boat.

You need to bring that to the other plugin developers. We only develop the radar plugin. None of the O developers reads this channel, I think.

sv-pizzazz, 2018-01-15:
I am only beginning to appreciate all the factors and discussions you have had and it is amazing. I believe everyone is having difficulties with the transition to a touch interface. When you look at the way the major vendors have implemented Touch on their chartplotters, it is really not optimal either. It is very inconvenient to adjust settings and this is why I still prefer an older chartplotter like the E-series to the new touch interfaces. OpenCPN is much better than any of them. 

I will think more about this and provide a more coherent and all inclusive proposal but I understand that you have been through these issues a few times. For now, my thoughts are:

1) Definitely no sliders - very difficult on a tablet on a moving boat. +/-1, 10 works well, it is used on time tested autopilots and it is very quick. I would even say that in most settings, a simple +/- with a step of 3 would work for most people. No one wants to adjust gain so precisely, speed of operation is more important.  

2) I see that in the new versions you have made the boxes all the same size, which is much easier than having them resize dynamically based on the text contents. This is what I meant by a fixed rectangular window, something that does not change all the time, so you know where to place the fingers. I also see the point in having a vertical layout, it takes less space on a typically horizontal tablet.

3) May be it is as simple as just re-ordering the nested menus based on the frequency of use. I believe there will be many opinions here and the right way is to ask people who actually use the radar often. To me, the key buttons are range (perfectly positioned) and gain controls (gain, sea, rain, FTC). One of my best experiences has been with the old Pathfinder radars where you press the button for Gain and you have the four gain controls appear at the bottom and you can adjust them all at the same time. This is how the military interface works as well. All the other functions can be nested. But then, some people will say, I leave in on Auto Gain, but I want the MARPA controls on the main menu. I see the problem. 

Thank you for the patience in bringing new users on board. I have shown this to at least 15 boaters on my marina over the weekend and they love it but had never heard of it before. At least I am helping popularize it

canboat, 2018-01-15:
There is even more for you to play with. If you keep the PPI windows open you can try working with the Control Menu Auto Hide setting found in preferences. I actually like to do that even though it means clicking on menu each time you want to do an adjustment, but I find the absence of the controls to be nice when just looking at the screen.

What we could do is that a recent addition to the PPI window is showing the gain / sea / rain clutter in the right hand bottom, with color coding for Off and Auto. We could make a click there make the radar control menu come up with that menu preselected (with that I mean the “Adjust” menu, as you’d have to be able to click quite delicately if it also tried to guess whether you wanted gain, sea or rain…) this would save 1 click.

Hakansv, 2018-01-15:
About touch screens. 
I've a RPi with the little 7 inch touch screen and I find all the radar controls are very easy and precise to adjust. The +- 1; +-10 are well detached and big enough to hit using my (thick?) finger. If you find the +-1 unnecessary I think it would not be a big thing to simply not use them? 

Now we've no sliders here on the radar_pi but have you tried the sliders in OCPN for Android? They do work even on a small mobile.