opencpn-radar-pi/radar_pi #83: GarminxHD struct radar_line bytes length problem

Xeriou, 2018-02-09:
Hello, 

I think I found a bug... maybe?
I trying to port garminxhd receiver to python 
(for research reason, I don't need the other function)
during get the radar packet, I found a strange thing

if the data length is 731 bytes in python
according the c++ struct in [here](https://github.com/opencpn-radar-pi/radar_pi/blob/d46b766c1b8e92e2b1e1fa06cb6d696e6e3a0d6b/src/garminxhd/GarminxHDReceive.cpp#L53)
the line_data should be 696(734 - 38)  bytes in python

but, when I parse packet, the scan_length_bytes_s is 698,
so, I think, the last 

>  uint16_t fills_5;

is line_data too.

or, if there is something i missed, please help me...

trudK45, 2018-02-09:
I will look into this when I have the opportunity.
Right now I don't have a development environment since I had to reinstall windows after a disk crash.
Today we are sailing down to St. Lucia and I am not sure there will be good internet there.

trudK45, 2018-02-10:
I have now managed to set up my development environment again.
I suppose you are referring to the 2 Nm range since that is the one that has got 731 (0x2db) bytes in the package including user header of 36 bytes. In the header it says that the number of radar data is 0x2b7 (695) bytes. 695+36 =731. So all adds up the way it should and as it is in the code.
Now, where did you get the numbers 734 and 38 from ?
Have you traced the data with Wireshark ?

Xeriou, 2018-02-12:
Ahh !!...
Sorry... this is my mistake !!!
I add more two bytes ....
you can close this issue ...
-----
I have done my code
many thanks for your c++ code !!!!!

closed 2018-02-12 by Xeriou