opencpn-radar-pi/radar_pi #87: Oscillating ranges

douwefokkema, 2018-06-03:
When starting two computers with radar_pi and they come up on different chart scales, conflicting auto-range commands will be send to the radar. This causes the radar to oscillate indefinitely between the two ranges.
Possible solutions:
- Send the auto-range command only once and accept a different range 
- Don't send a range command at start-up. Radar will start at last range before shut-down

keesverruijt, 2018-06-03:
The first solution sounds ok.

The second solution either doesn't work or your analysis is incorrect. If
the problem only happens when *starting* O on 2 computers with different
chart scales,  then it would not keep oscillating as the command is only
sent at start-up. But you say you see oscillation, which means that both
instances of O keep sending range commands. Ergo, not sending something at
start-up is not enough to fix the issue.

douwefokkema, 2018-06-03:
At start-up pi is in auto-range.
When pi is in auto-range it will continuously compare actual range with chart scale (must be done in case user changes chart scale). If no fit, it will issue a range command. If 2 computers in auto-range and different chart scale, they will compete for the range and cause oscillation of range resulting in loss of radar image.

douwefokkema, 2018-06-04:
Solved in branch autorange. Just an ordinary bug. Still to be tested with real radar.