opencpn-radar-pi/radar_pi #88: Radar can not be switched to standby

douwefokkema, 2018-06-03:
Radar can not be switched to standby if radar is also displayed on another computer that was previously shut down with running radar.
Possible solution: Don't auto start radar after OpenCPN was closed with radar on Tx

keesverruijt, 2018-06-03:
I do not understand what you are saying exactly. The "other computer", is
it on or off? And is O running on it or not? If it is off or O is not
running then it is no longer sending commands to the radar, so there cannot
be any issue. If it is on, why does it matter whether it was previously
shut down?

douwefokkema, 2018-06-03:
On computer 1 O was closed with running radar. Then on computer 1 O is restarted but not touched anymore. Computer 1 automatically switches the radar to transmit. A user starts O on computer 2, gets a transmitting radar. User clicks button to switch radar to standby. Radar goes to standby, but immediately there after to spinning-up and transmitting again. User on computer 2 has no way to switch radar off permanently.

closed 2019-04-01 by keesverruijt