opencpn-radar-pi/radar_pi #89: Missing spokes after setting pixels per spoke to 1024 for Navico

douwefokkema, 2018-06-04:
Håkan reports incidental missing spokes in the highres implementation (branch highres). Currently a lot of work is carried out under control of the receive thread. All vertices for the image are prepared on a per spoke basis from the receive thread. Doubling the numbers of pixels per thread might make this critical on slower machines, resulting in loss of packets.
In branch highres commit 136584d07 the priority of the receive thread is set to wxPRIORITY_MAX. Needs to be tested if this solves the issue.

keesverruijt, 2018-06-04:
I thought that Hakan reported back that it was related to WiFi?

douwefokkema, 2018-06-04:
Yes, initially. But on the wired connection there still seems to be some intermittent packet loss. As I did not change the network traffic but increased the processing in the receive thread I just try the above solution (increase prio). Better would be to get all the processing out of the receive thread, so separate ProcessRadarSpoke from the receive thread. Receive thread could dump all spokes in a FIFO buffer which is processed in a new async thread. But that is quite some job and should be well designed not to harm the architecture of the system.

douwefokkema, 2018-06-05:
tested OK. Merged.

closed 2018-06-05 by douwefokkema