opencpn-radar-pi/radar_pi #91: Recording raw radar data?

ElOceanografo, 2018-08-23:
A feature request: How difficult would it be to add an option to the plugin allowing the user to record raw radar output (i.e. an array of reflectivity and/or radial velocity for each antenna rotation)?

For context, I'm a biologist who has used marine radar to study seabird behavior.  I've previously written my own code to control an ADC hooked into the radar, capturing raw data that way.  With newer networked digital radars that approach should be unnecessary, and it seems like this plugin already has nearly all the necessary functionality, at least in terms of reading and decoding data off the network.

I recognize this feature may be of limited interest to most boaters, so I'd be willing to take it on myself if someone can point me towards the right spots in the codebase to get started...

Hakansv, 2018-08-24:
@ElOceanografo 
First I suggest you read one of the basic paper for this project: http://www.roboat.at/technologie/radar/
Study the pdf document linked to on the site for details.
Then read the radar_pi code here how distance, angle and strength for each blob is received and printed on screen and you may find a way to collect the data you need.
I'm not sure of your intentions but the existing radar_pi ARPA function may be a way to follow a bird track and also record it by the OCPN VDR plugin?

douwefokkema, 2018-08-24:
Welcome on board! 
Fine you use already the new radar_pi, it offers the double line resolution compared with the original br24radar_pi (1024 pixels per line instead of 512). 
I have been looking at logging radar data. I thing it is a good thing to be able to replay your voyage, including radar image, both for learning and safety. On commercial radars it is a requirement to record the recent hour (?) of the trip in case of accidents. And our radar should meet commercial (IMO) standards. 
An easy way to log radar data is to use Wireshark. It will record all network traffic,  you can set it up to only record the radar IP. If you output the GPS data simultanously on UDP, you get the full picture including position. This all can be replayed with tcpreplay (Unix) , offering one to one correspondence  with the original situation. But Wireshark files get very large, they can be zipped, but only after recording them. So it is not practical to do continuous recording with Wireshark. May be under Unix on the fly compression is possible.
When recording under program control further compression can be done. A radar return (intensity) has a size of 4 bits. But normally is is 0 or high. So a single bit could be sufficient to record a return (unless you need the intensity for birds), 0 if below threshold, 1 above threshold. Combining multiple pixels in a single number and not recording identical lines next sweep very high compression rates can be achieved, depending on how "filled" the image is.
You can reach me directly at df@percussion.nl
I was thinking packaging these compressed lines into pseudo NMEA messages and get them recorded by the VDR (voyage data recorder) plugin. Then VDR would take care of the logging mechanism including positon and others, including replay. That would leave me with the task to make intercept the replayed "NMEA radar" messages and put them on screen. But here I got stuck. The NMEA handler does not accept  binary messages. Certain characters have special purpose and may not be used. I stopped there.
I do have a branch with these trials: https://github.com/douwefokkema/radar_pi branch vdr.
my mai

sv-pizzazz, 2021-11-22:
My the NMEA RSD sentence can be helpful here. It is a standard NMEA sentence sent and received by Raymarine radars for sending target returns from one screen to another. I think it is a leftover from the Raytheon days when one ship would detect a target and pass it over to another ship in the fleet. Would it be possible to generate such a nmea 0183 sentence after the ARPA routine and be saved in the VDR? At some point I was contemplating using the RSD sentence to send AIS data to the Raymarine RL70 displays that have excellent reliability, chart and radar overlay but do not understand AIS. However, I never wrote the code for that but I can confirm that you can pass targets in this way from an old Raymarine RL70 analog radar to any Raymarine display.

 RSD - RADAR System Data
                                                        14
        1   2   3   4   5   6   7   8   9   10  11 12 13|
        |   |   |   |   |   |   |   |   |   |   |   | | |
 $--RSD,x.x,x.x,x.x,x.x,x.x,x.x,x.x,x.x,x.x,x.x,x.x,a,a*hh<CR><LF>

 Field Number: 
  9) Cursor Range From Own Ship
 10) Cursor Bearing Degrees Clockwise From Zero
 11) Range Scale
 12) Range Units
 14) Checksum

keesverruijt, 2021-11-22:
The RSD sentence is processed information, and since it represents only a cursor range and bearing from the ship it is nothing more than the NMEA0183 representation of the EBL/VRM (Electronic Bearing Line/Virtual Range Marker) functionality of a radar.

mohanrobotics, 2022-07-19:
I want to record the radar data from Halo24 radar. Any inputs on this will be really helpful.

Hakansv, 2022-07-19:
@mohanrobotics ..
Apart from what's mentioned above there are no other hints. Well, you can always capture a movie of what's on screen if that's more convenient?

mohanrobotics, 2022-07-19:
@Hakansv Thanks for the reply. But i want the data to perform some analysis. And the link you mentioned about the paper doesn't seem to work. My understanding is that this plugin receive the data as bytes through UDP, parse the data, perform some calculations and display it. I am looking through the code currently and it will be really helpful where that PARSED data is generated.