opencpn-radar-pi/radar_pi #92: support for OpenCPN 5.0 features (multicanvas / MUI)

LeaseOnLife, 2018-09-12:
I am running successfully the current master branch (4.99) of OpenCPN with the radar_pi (B&G 4G). I switched to the development version because of support for charts in mbtiles format (which I export using SAS PLANET).

There are more upcoming features, I really like multicanvas support, displaying two charts side by side. Even better is the branch "MUI" featuring a new 'flatter' GUI, see screenshot.
[image]

When trying to use radar_pi with the multicanvas or MUI branch of opencpn, it crashes almost instantly after starting OpenCPN.

Beta for opencpn 5.0 is still a few months out, per Dave, but I wanted to make the dev team here aware of work needed to support 5.0. 

I am happy to test!

Dirk

keesverruijt, 2018-09-23:
Hi Dirk,

Thanks for the pointer. I was busy sailing & planning a refit for the boat so I haven't been very actively looking at 5.0 yet but I will take on the job of making sure the plugin works with the MUI branch.

keesverruijt, 2018-10-08:
With the 4.99.1007 version of today's MUI branch it actually works surprisingly well. There are still a number of issues but the crashing seems to have gone, and I get radar overlay on both chart canvases.

LeaseOnLife, 2018-10-12:
just got the latest from git, emptied the plugin directory, compiled opencpn (shows as 4.99.1009) and radar_pi. For me this still crashes. It does not crash when libradar_pi.so is removed, or when opencpn is set to only show one canvas.

Hakansv, 2018-12-04:
@keesverruijt  @douwefokkema 
Hi. I think it's due time to start investigate radar_pi collaboration by O V.5 and the MUI branch of OpenCPN.
I assume the MUI branch is not far from a beta test period. Most is now working fine and the ongoing tests are more or less fine tuning.
There are two possible modes to run O. In one Canvas and two Canvas mode. 
In one Canvas mode I can't find any big obstacles for radar_pi. It's working fine except the two issues mentioned further down.
In two Canvas mode and for overlay plugins like grib_pi and radar_pi the only available Canvas is, so far, the right one. Radar_pi is working as expected also there.
For radar_pi I think we have to:
1. Investigate if we, together with Dave, in some way can support an O API to give availability also to the left Canvas. That is to direct an overlay to a specific Canvas in the case both are used. As mentioned this would be useful when two radars are available as for the 4G.
2. Check if the O crash when AIS targets is used in the PPI window is related to O-MUI or if it is a radar_pi specific issue.
3. Check if the PPI docking issue is MUI related or radar_pi specific; When the PPI window is docked, intentional or by accident, there is no obvious way to undock. I've to "hack" the config file's AUI section. 

I'm, as you know, not the experienced coder to dig deep into these tasks but whatever I can support it would be a joy. Please advice.
Håkan

[image]

Hakansv, 2018-12-18:
This task:

> Check if the O crash when AIS targets is used in the PPI window is related to O-MUI or if it is a radar_pi specific issue.

It's now solved by Douwe and PR to OCPN branch MUI. It was related to the MUI update changes and not to radar_pi