opencpn-radar-pi/radar_pi #93: Clarifying radar type Navico BR25 and 3G

Hakansv, 2018-09-24:
To make it easier for a user to make correct choice I suggest text change as of the pict below. 3G using firmware before 2014 shall act as a RT_BR24 otherwise as RT_3G.
Will the string be to long string and extent the radar control window to much? Opinions please. If not I'll merge the change. 

[image]

keesverruijt, 2018-09-24:
I think we should put this in the documentation only. My guess is there are not so many people with old 3G as the price difference is/was not high. I will ask Navico how the distribution 3G/4G is.

If we do put up a text it should be something like ‘Navico BR24 and 3G < 2014’.

Kees

> 
> To make it easier for a user to make correct choice I suggest text change as of the pict below. 3G using firmware before 2014 shall act as a RT_BR24 otherwise as RT_3G.
> Will the string be to long and extent the radar control window to much? Opinions please. If not I'll merge the change.
> 
>  <https://user-images.githubusercontent.com/7202854/45958554-0034ab80-c018-11e8-97f2-3c68bbe0c630.png>

Kind regards,

Kees Verruijt
—

Kees Verruijt
Midlumerlaan 67
8861 JJ  Harlingen
The Netherlands

Cell +31 6 24694455
Home +31 517 854475

Hakansv, 2018-09-24:
I'm not sure at all and it's not tested. It was based on a comment on Lowrance that firmware after 2014 would need an update for the HDS but that could be other reasons like extended range. Your contacts are for sure better than my imagination. Let's scrap it and maybe mention it as an uncertainty in the manual. 
Thanks /Håkan

closed 2018-09-24 by Hakansv