opencpn-radar-pi/radar_pi #97: commits on and past 12/30 crash opencpn

LeaseOnLife, 2019-01-06:
@douwefokkema: I tried your commits and today's bundle crashes my opencpn.  (Dell E7470, Ubuntu 18.04, B&G 4G).
Tracing back commit by commit, I can successfully run with commits before 12/30/2018. With commits on and past 12/30 it crashes. I have not tried all commits in between and currently have not enabled debug. 

Let me know if I can be of further help.
Dirk

douwefokkema, 2019-01-07:
Hi Dirk,
30 December the multi canvas buttons were added. I checked the code and made the button handling more secure, now on radar_pi master. But I do not know if this solves the issue. Please test. I do not have a working Linux development environment at hand. If still crashes, try to make a log file with verbose=31 (in the. ini file, .conf on Linux?). Better even if you could compile with debug. Thanks for your help. 
Douwe

LeaseOnLife, 2019-01-07:
Hi Doug,
with your latest commits and VerboseLog=31 in the radar section of the config file, it still crashes, not sure the log will give you anything meaningful:
==========================================================
11:45:14 EST: PlugInManager: Loading PlugIn: /usr/local/lib/opencpn/libradar_pi.so
11:45:14 EST: PlugInManager:  /usr/local/lib/opencpn/libradar_pi.so
              API Version detected: 116
              PlugIn Version detected: 1010
11:45:14 EST: radar_pi: LoadConfig: show_radar[0]=0 control=819,132
11:45:14 EST: radar_pi: Configuration file values initialised
11:45:14 EST: radar_pi: Log verbosity = 31. To modify, set VerboseLog to sum of:
11:45:14 EST: radar_pi: VERBOSE  = 1
11:45:14 EST: radar_pi: DIALOG   = 2
11:45:14 EST: radar_pi: TRANSMIT = 4
11:45:14 EST: radar_pi: RECEIVE  = 8
11:45:14 EST: radar_pi: GUARD    = 16
11:45:14 EST: radar_pi: ARPA     = 32
11:45:14 EST: radar_pi: VERBOSE  log is enabled
11:45:14 EST: radar_pi: DIALOG   log is enabled
11:45:14 EST: radar_pi: TRANSMIT log is enabled
11:45:14 EST: radar_pi: RECEIVE  log is enabled
11:45:14 EST: radar_pi: GUARD    log is enabled
11:45:14 EST: radar_pi: Initialized plugin transmit=0/0 
11:45:14 EST: radar_pi: Added panel radar_pi-0 to AUI control manager
11:45:14 EST: radar_pi: Target trail value 0 = 0 revolutions
11:45:14 EST: radar_pi: Navico 4G A starting receive thread
11:45:14 EST: radar_pi: Navico 4G A receive thread created, prio= 100
11:45:14 EST: radar_pi: Radar B GuardZone 1: destroyed
11:45:14 EST: radar_pi: Radar B GuardZone 2: destroyed
11:45:14 EST: radar_pi: Radar C GuardZone 1: destroyed
11:45:14 EST: radar_pi: Radar C GuardZone 2: destroyed
11:45:14 EST: radar_pi: Radar D GuardZone 1: destroyed
11:45:14 EST: radar_pi: Radar D GuardZone 2: destroyed
11:45:14 EST: radar_pi: RadarWindow[0] show=0 showcontrol=0
11:45:14 EST: radar_pi Navico 4G A: set visible 0
11:45:14 EST: radar_pi: ContextMenu arpa nr of targets = 0
11:45:14 EST: radar_pi: ContextMenu show = 0
11:45:14 EST: radar_pi: messagebox decision: show=0 overlay0=-1 overlay0=-1 auto_hide=1 opengl=0 radarOn=0 navOn=0
11:45:14 EST: radar_pi: messagebox no radar wanted: HIDE
11:45:14 EST: radar_pi: no change
11:45:14 EST: PlugInManager: Radar has loaded very slowly: 67 ms
11:45:14 EST: Opencpn language load for:  en_US
11:45:14 EST: Loading catalog for:  opencpn-climatology_pi
11:45:14 EST: Loading catalog for:  opencpn-grib_pi
11:45:14 EST: Loading catalog for:  opencpn-wmm_pi
11:45:14 EST: Loading catalog for:  opencpn-chartdldr_pi
11:45:14 EST: Loading catalog for:  opencpn-radar_pi
11:45:14 EST: Loading catalog for opencpn core.
11:45:14 EST: Getting DefaultSystemLocale...
11:45:14 EST: PlugInManager searching for PlugIns in location /usr/lib/opencpn
11:45:14 EST:    Directory /usr/lib/opencpn does not exist.
11:45:14 EST:    ***GPS Watchdog timeout at Lat:9.367   Lon: -79.951
11:45:14 EST: radar_pi: WMM plugin provides new magnetic variation -4.374834
11:45:15 EST: radar_pi: SetPositionFixEx var=nan var_wd=1
11:45:15 EST: TCP Datastream connection established: TCP:127.0.0.1:10110
11:45:14 EST: radar_pi: NavicoReceive thread Navico 4G A starting
11:45:14 EST: radar_pi: multicast reception for 236.6.7.9 port 6679 on interface 1.1.1.100
11:45:14 EST: radar_pi: Navico 4G A scanning interface 1.1.1.100 for data from 236.6.7.9 port 6679
11:45:15 EST: radar_pi: select maxFd=11 r=0 elapsed=250
11:45:16 EST: radar_pi: RenderGLOverlay context=0x5585da42c080
11:45:16 EST: radar_pi: RadarWindow[0] show=0 showcontrol=0
11:45:16 EST: radar_pi Navico 4G A: set visible 0
11:45:16 EST: radar_pi: messagebox decision: show=0 overlay0=-1 overlay0=-1 auto_hide=1 opengl=1 radarOn=0 navOn=0
11:45:16 EST: radar_pi: messagebox no radar wanted: HIDE
11:45:16 EST: radar_pi: no change
11:45:16 EST: radar_pi: rendering PPI window(s) took 0ms, next extra render is in 250ms

douwefokkema, 2019-01-09:
Crash should be corrected in last commit "Solution for crash on Linux", ed27aa. Please test.
Douwe

LeaseOnLife, 2019-01-09:
That fixed it, thanks much! issue closed!

closed 2019-01-09 by LeaseOnLife