opencpn-radar-pi/radar_pi #99: Build errors on windows

trudK45, 2019-01-25:
I am trying to build with latest but get a lot of link errors. A total of 69 errors.

First being:
NavicoControl.obj : error LNK2001: unresolved external symbol "public: virtual __thiscall opencpn_plugin_116::~opencpn_plugin_116(void)" (??1opencpn_plugin_116@@UAE@XZ) [D:\transfer\Program\OpenCPN_program_and_plugins\radar_pi\build-win32\radar_pi.vcxproj]

With code 4.610 I have no problems building radar_pi.

/Hans

Hakansv, 2019-01-25:
Hi Hasse.
Yes, we, most Douwe and Kees, are close to finish to fit to O V5.0 candidate. Ocpn master is close to beta for Version 5 and radar_pi master is now adapted to that. O V5 will be great with what's called multi canvas. That is we can split the screens in two and show the boat symbol independently. Radar_pi will do the same, For a 4G two "radars" can be used at the same time.
To build now we need to use wxWidgets 3.1.1 in Windows. All described here: https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:developer_manual:developer_guide:compiling_windows
You can build O master and then radar_pi master using the new opencpn.lib.
For the very moment, since yesterday, there's a change in O API development to adapt to other plugins so radar_pi will not use multi canvas in a good way. That will be fixed in a couple of days I suppose.  But If you'll try the new radar_pi code it will still work fine using single canvas mode in O V5.

How's life "over there". 
Hälsningar
Håkan

trudK45, 2019-01-25:
That is the document I have used and tried the "Compile on Windows (to OpenCPN 4.8.8)"
I am using VS 2013, wxWidgets-3.0.2.
The opencpb.lib created is 138 KB.

Life is good. We are on Guadeloupe and soon heading for Antigua.

/Hans

Hakansv, 2019-01-25:
OK but our environment is now changed to VS 2017 and wx3.1.1. You've to start from head of the page and not down the "old" description for 4.8.8.
Sorry but development is the word that's matter I think. And OpenCPN is not standing still. Dave, and others, are for sure productive. Here's a shot. The mid bar can be dragged to any position and scales changed independently. For the moment not all plugins are adapted. If you're interested to try tell me what plugins you're using and I can give actual status for O V5.
[image]

trudK45, 2019-01-25:
[image]

trudK45, 2019-01-25:
OK. I will wait until these plugins are ready and use an older radar_pi plugin in the meantime.
The drawback will be I can not do any development on radar_pi unless I make a parallel development environment.

/Hans

Hakansv, 2019-01-25:
Hasse.
I just cloned and built gecomapi_pi. 
Builds fine and seems to work also in V5 beta candidate.
When I moved the window a docking occurred I can't get rid of but I've never used that plugin so it might be solved.
[image]

JeroenAdam, 2019-02-22:
Hi, I got the courage to try building this plugin, I'm having the following build error.

  radar_pi.dir\Debug\NavicoReceive.obj
  radar_pi.dir\Debug\NavicoControl.obj
  radar_pi.dir\Debug\NavicoControlsDialog.obj
**LINK : fatal error LNK1104: cannot open file '..\wxWidgets\wxWidgets\lib.obj'** [C:\Users\admin\projects\br
24\radar_pi\build-win32\radar_pi.vcxproj]
Génération du projet "C:\Users\admin\projects\br24\radar_pi\build-win32\radar_pi.vcxproj" terminée (cibles par défaut)
-- ÉCHEC.

Génération du projet "C:\Users\admin\projects\br24\radar_pi\build-win32\ALL_BUILD.vcxproj" terminée (cibles par défaut)
 -- ÉCHEC.

ÉCHEC de la build.

"C:\Users\admin\projects\br24\radar_pi\build-win32\ALL_BUILD.vcxproj" (cible par défaut) (1) ->
"C:\Users\admin\projects\br24\radar_pi\build-win32\radar_pi.vcxproj" (cible par défaut) (3) ->
(Link cible) ->
  LINK : fatal error LNK1104: impossible d'ouvrir le fichier '..\wxWidgets\wxWidgets\lib.obj' [C:\Users\admin\projects\
br24\radar_pi\build-win32\radar_pi.vcxproj]

    0 Avertissement(s)
    1 Erreur(s)

Temps écoulé 00:01:37.60

Hakansv, 2019-02-22:
Hi, Jeroen.............
First of all you must use wxWidgets 3.1.2 and follow, exactly, what Pavel has wrote here: https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:developer_manual:developer_guide:compiling_windows
I suppose you've read what's said above in this issue? (Yes, it's changed from wx3.1.1 to wx3.1.2)
Håkan

keesverruijt, 2019-04-01:
At the moment until we update the doc the best recipe to follow is the `appveyor.yml` file in the home directory as that is what we use to build the Microsoft Windows continuous integration version.

closed 2019-07-21 by keesverruijt