opencpn-radar-pi/radar_pi #107: Navico HALO3 detection is not reliable

keesverruijt, 2019-04-07:
Eric reports that the new HALO/4G detection code that works for HALO24 does not work for HALO3.

We will need PCAP files of:

- Power on of radar and no OpenCPN nor chart plotter. Please capture 120s starting from power on. 
- Power on of radar and chart plotter, with chartplotter showing it recognised radar. No need to turn radar "on" (e.g. transmitting). Please capture 120s starting from power on.

If you can also add the following two files:

- Power on of radar, plotter and OpenCPN, with plugin *not detecting* radar.
- Power on of radar, plotter and OpenCPN, with plugin *detecting* radar. 

Please make these into separate files with names that explain what they contain. 
You can attach the files to this issue.

pi-gizmos, 2019-04-09:
I cannot exactly do the power on of the MFD as it controls the power on of the router which takes longer to power up.  I tried to simulate it in item 4 by turning on the radar and plugging the ethernet into the MFD at the same time.  If there is another way let me know, it is very difficult to get to the router to change the power on wire.  The filenames should explain what I did.
Best, Eric
[Halo3_turn-on.zip](https://github.com/opencpn-radar-pi/radar_pi/files/3056760/Halo3_turn-on.zip)

keesverruijt, 2019-04-10:
Hi Eric, thanks. I will need a few days to look into this.

keesverruijt, 2019-07-21:
@pi-gizmos are you still experiencing issues with the latest beta 5.0.3-beta2?

pi-gizmos, 2019-07-22:
Hello,

I just downloaded it and tried it for  bit.  It does not have any more
issues coming up with the correct "scanning 192.168.77..." text box
reliably as it did before.  It does not always connect to the scanner, it
just blinks scanning the interface, but I think the issue may be my WiFi
adapter.  It happens after I switch APs a few times (from my iridium Go to
my internet hotspot to my nav wifi)  It is always solved by rebooting the
computer.  When I select the nav WiFi first after reboot OpenCPN always
connects to the radar and I cannot make it fail.  I will look into updating
my WiFi adapter driver, I think the problem is now mine not yours.

It does however, now have data blackouts.  There are between 3-6 missing
sectors per revolution now, that I never saw before.  They are different
sizes, I could send a screenshot, but I think you know what they look like
very well.

Thanks for all of your effort. If there is any data I can send to help
(wiresharks etc) let me know.

Eric

keesverruijt, 2019-07-22:
Missing sectors is to be expected with WiFi. We do not recommend WiFi but a physical LAN connection.

> On 22 Jul 2019, at 06:16, pi-gizmos <notifications@github.com> wrote:
> 
> I just downloaded it and tried it for bit. It does not have any more
> issues coming up with the correct "scanning 192.168.77..." text box
> reliably as it did before. It does not always connect to the scanner, it
> just blinks scanning the interface, but I think the issue may be my WiFi
> adapter. It happens after I switch APs a few times (from my iridium Go to
> my internet hotspot to my nav wifi) It is always solved by rebooting the
> computer. When I select the nav WiFi first after reboot OpenCPN always
> connects to the radar and I cannot make it fail. I will look into updating
> my WiFi adapter driver, I think the problem is now mine not yours.
> 
> It does however, now have data blackouts. There are between 3-6 missing
> sectors per revolution now, that I never saw before. They are different
> sizes, I could send a screenshot, but I think you know what they look like
> very well.
> 
> Thanks for all of your effort. If there is any data I can send to help
> (wiresharks etc) let me know.
> 
> Eric

douwefokkema, 2019-11-29:
This issue should be solved with v5.0.4-beta6. Please test with your radar.

pi-gizmos, 2019-11-29:
Hello,

Thanks much for the fix. Using Wifi  I was usually having a few missing
sectors with the old version, they didn't make the radar unusable, but they
were noticeable.  Now they are almost completely gone.  They are completely
gone at the normal offshore scan rate.  There is only one thin missing
sector, maybe once every ten seconds or so at the double (harbor mode) scan
rate.  If I wasn't looking for them, I would likely not even notice. So it
is vastly improved.  Thanks also for changing the +/- buttons that change
the zoom level, it now matches the chart plotter.  I didn't find
anything else unusual during 1/2 hour of testing, everything works as
expected.  I'll keep using and let you know, but otherwise it works great.

Best, Eric
SV Scoots
Currently Opua, New Zealand

douwefokkema, 2019-12-01:
My update was primarily meant to solve the Halo detection problem. If it solves WiFi issues as well, fine. I will close this issue for now, we can re-open if needed.

closed 2019-12-01 by douwefokkema

pi-gizmos, 2019-12-01:
Yes,
Halo Detection is working very well.  I have an issue if I switch to the
radar network  after a long time on the other, internet connected network.
 I believe this is a windows DNS cache or firewall issue.  It is always
resolved by rebooting.  The radar is always connected perfectly after the
reboot.  From my perspective you can close this issue.
Thanks for your work on this.

Best, Eric